prusa3d/Prusa-Firmware-Buddy #3535: [BUG] XL Stops mid print without appearent reason

Mark-Oberst, 2023-11-28:
**Printer type** - [XL]

**Printer firmware version** - 5.1.0

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None. Stock Prusa XL]

**USB drive or USB/Octoprint**
Print using USB Drive

**Describe the bug**
I started printing off a large batch of parts. The XL stopped mid-print during printing with no error or anything. I could hear a very faint clicking sound. Not nearly as loud as a stepper skipping steps, but still clicking. Also, quite unusual that the print stopped without moving the print head away from the print or cooling it down in any way. I also couldn't "pause" the print. It would not react to that. I could navigate through the menu, but the options didn't do anything.
I tried "recovering" the print using the power loss recovery feature. That seemed to work, the printer rehomed and was about to start printing again. It then moved the printhead to the exact spot where it left off. There, it stayed and was stuck again.

I measured the parts and they were 4mm high - so it seems to have stopped printing after 20 layers.

**How to reproduce**
I do not rerun the gcode because I need the filament. I have however attached the gcode and 3mf files used for that print.

**Expected behavior**
Keep printing or at least give an error when stopping.

**G-code**
[Gcode and 3mf.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13486803/Gcode.and.3mf.zip)

**Pictures**
[pictures.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13486833/pictures.zip)

Mark-Oberst, 2023-11-28:
Ok, so I just reran part of that print. I excluded the part where it got stuck and some other parts. I also changed some of the print settings like extrusion width, but nothing major. 
At some point, I started hearing an unexpected sound - like the Z-Axis moving all the way down so I went to look, and I can only describe what the printer was doing as self-destruct mode.

I caught it on video. It was first moving the bed all the way down and then the X and Y axes. The Toolhead basically tried to behead itself. See video.

I read in the patch notes that crash detection is disabled for the XL. This might be a safety problem with this bug.
I had my thumb on the reset button and I'm not totally sure if I was the one hitting the reset button the last second or if some printer safety feature triggered. Is there even a safety feature that would have been triggered? This is quite concerning.

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/145592853/5454ade1-d130-4384-bc5d-f86f37a0375c

gcode i was printing: 
[gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13493267/gcode.zip)

Error occured 1.6mm into the print while printing the following part: 

[image]

JeremiahGillis, 2024-02-27:
I just had mine stop right in the middle of purging the first nozzle at the beginning of my print. I was able to recover by pausing and unpausing. I've seen this multiple times. Hopefully someone is working on this since it has been three months now. I am on v5.1.2.

closed 2024-06-27 by github-actions[bot]