prusa3d/Prusa-Firmware-Buddy #3536: Show message during M600 filament change

magnushacker, 2023-11-28:
When doing multi-color prints using the manual method (e.g not via MMU, but via M600 filament change commands), it would be good if the printer screen can tell me which filament to insert. This could be done by assigning each extruder a nickname in the slicer and having this name show during the M600 change process (e.g "Changing to BLUE PLA" instead of the default texts on the screen).

bkerler, 2023-12-12:
I saw that the firmware is already prepared to show a color but it hasn't been fully implemented in the firmware. So it would be possible to add a "C" flag with the color as integer value, such as "M600 X .. Y .. C 65280". Color value is defined as "red << 16 | green << 8 | blue", whereas 65280 should be pure green in this case.

bkerler, 2023-12-12:
I just created the PR #3581 for adding the color option to M600. Not sure yet how the filament type is being shown / handled yet.

bkerler, 2023-12-12:
Regarding the filament type, this should be automatically be shown on filament change according to this line:

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/15d5ae0b317a37b69b8e7a4720711b962c25b57d/src/marlin_stubs/pause/M600.cpp#L204

I will dig deeper to see where the filament type is being set into the target_extruder value

magnushacker, 2023-12-12:
My idea was to have a text option to M600 which could be used to give a text to show. My current approach is to have a pause, then when I choose to resume I display a text with the M117 command for 4 seconds and then do the M600 command. There are a couple of problems with this approach; first, the printer performs some unnecessary movements for the pause and I have to choose Resume from the menu. Second, it's not super user friendly to simply say "next color is FF0000" instead of "next color is red" (I only have the extruder color available in the gcode). Also, there is a risk I forget which color I was supposed to change to once the unload sequence in M600 is finished :)

Of course, having some other text/description associated with each virtual extruder will also require changes in the slicer.

magnushacker, 2023-12-12:
Any text displayed with M117 will get erased by the texts showing by the M600 command, hence the need for a pause before the M600 is executed.

bkerler, 2023-12-12:
Normally you don't set the color manually but you use the prusaslicer and it would set the color in the gcode according to the extruder settings (that's how it's currently implemented in the firmware). I doubt regular users write the gcode by themselves :) Adding text colors would need a table for the colors, that would surely blow up firmware size unnecessary.

Just showing text on M600 is harder to implement but doable, but I can't test that right now without an additional xbuddy board as the board is currently sold out.

bkerler, 2023-12-12:
So do I understand you right that the behaviour should be the same, but except for the "Load filament" message on the screen a custom message should be printed ?

bkerler, 2023-12-12:
I think replacing the hardcoded text "Insert filament and press button" is pretty hard as it is hardcoded deeply. But it should be possible / a lot easier to show a message after the button has been pressed instead, similar to M117.

magnushacker, 2023-12-13:
> So do I understand you right that the behaviour should be the same, but except for the "Load filament" message on the screen a custom message should be printed ?
> 
> The current behaviour is:
> 
> pause the print (pause_print) show_continue_prompt (beep before), show msg "Insert filament and press button" <<< replace text here ? Wait for filament insert by user and press button, show msg "Nozzle parked" resume the print (resume_print) show message ("Load filament")

Yes, correct. Changing the next message ("Load filament") wouldn't help I think since at that point you have already loaded the new filament, right?

My current tool change gcode looks like this:

```
M601 ; Pause
M400 ; wait until all moves are completed
; color notification
M117 Next color is { (extruder_colour[next_extruder] == "#00FF00" ? "GREEN"
:(extruder_colour[next_extruder] == "#FF0000" ? "RED"
:(extruder_colour[next_extruder] == "#FFFF00" ? "YELLOW"
:(extruder_colour[next_extruder] == "#0000FF" ? "BLUE"
:(extruder_colour[next_extruder] == "#FFFFFF" ? "WHITE"
:(extruder_colour[next_extruder] == "#000000" ? "BLACK"
:(extruder_colour[next_extruder] == "#999999" ? "GRAY"
:(extruder_colour[next_extruder] == "#D4AF37" ? "GOLD"
:(extruder_colour[next_extruder] == "#C0C0C0" ? "SILVER"
:(extruder_colour[next_extruder] == "#FF8000" ? "ORANGE"
:(extruder_colour[next_extruder] == "#800080" ? "PURPLE"
:(extruder_colour[next_extruder] == "#FF6600" ? "ORANGE"
:(extruder_colour[next_extruder] == "#F0F0F0" ? "TRANSPARENT"
:(extruder_colour[next_extruder] == "#800000" ? "BROWN"
:extruder_colour[next_extruder] ))))))))))))))};
G4 S4 ; wait 4 seconds
M600

```
Ideally, I'd like to replace this with something like this:

`M600 T Change to: { extruder_nickname[next_extruder] } `

but as we don't have the extruder_nickname set in the slicer yet, the next best thing would be:

`M600 T Change to: { extruder_colour[next_extruder] } `

(possibly with a huge IF statement like the one I use today)

bkerler, 2023-12-13:
I will try and take the colors you are referring to and add them to my current PR so that no integer value has to be used. The biggest problem with the custom messages is that the gcode itself is handled independently from the nextruder (which makes sense for various reasons).

bkerler, 2023-12-13:
PR #3581 has been adopted. Now you can use "M600 C RED S PLA" or "M600 C 16711680 S PLA" as well.
I've added these colors :
    BLACK,
    BLUE,
    GREEN,
    BROWN,
    PURPLE,
    GRAY,
    TERRACOTTA,
    SILVER,
    GOLD,
    RED,
    PINK,
    ORANGE,
    TRANSPARENT,
    YELLOW,
    WHITE

Regarding the custom message, that will be a different PR but I need to get my dev board first.

bkerler, 2024-02-20:
Maybe showing a message without having to use M600 would be via `M0 msg`, see this PR #3779

danopernis, 2024-06-05:
This should be fixed since v6.0.0 via commit https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/b80e750a21b248535770536cbb5a9db079f42eef

Can we close this?

closed 2025-02-10 by github-actions[bot]

Tarakesh83, 2025-05-02:
When can we expect that #3581 makes it to released firmware?
On Feb. 10 it was stated that it would be tracked down why this is not implemented yet.
Working with workarounds adds massive amounts of time just to be able to read what color is coming up next when doing manual changes.
For  example 200 canges times 4 seconds  wait time amounts to  nearly 15 minues of the printer doing absolutly nothing.
plus the moves for pause and unpause and this adds up to hours of extra time depending on the  amount of filamen changes