prusa3d/Prusa-Firmware-Buddy #3537: [FEATURE REQUEST] Do not go back to last position after pause

SimonTil, 2023-11-28:
[MINI] - though, had the same issue with my MK2S)

When a pause or color change is executed through an M600 command, the nozzle moves back to the last position causing a little blob of the new color on the old position. The expected behaviour is either first prime the nozzle (in a prime tower, or in the skirt / draft shield), or go toward the new position immediately after the pause.
It's not in the g-code (otherwise I would already have fixed the issue manually), so, has to be a firmware thing. That's why issued here.

This is a really weird behaviour. Why would the nozzle even move to the last position, if it should stop there to go on with another color on the next position?

MichaelJLew, 2023-11-28:
I have this same issue and think that it is a bug rather than a feature request.

MichaelJLew, 2023-11-28:
I should have said that I have this same issue _on my MK4_.

SimonTil, 2023-11-28:
> I have this same issue and think that it is a bug rather than a feature request.

I don't think so, because to come up with a solution, a decision has to be made:
- either move back to the original position (as it is now)
- or only move to the pause position and put an additional command after the M600 to go to the next position in the g-code (through PrusaSlicer).

And that's why I request this as a feature request. Because you can't expect other slicers to do the same as PrusaSlicer (see seconde option), though, you should be able to use other slicers with your Prusa Printer.

MichaelJLew, 2023-11-28:
I'm not sure that I see your point. The firmware knows both the last point to which it moved with extrusion and the next post pause point that includes a move with extrusion. 

Surely after the colour change it should go to that next point no matter which slicer was used. If the slicer wants one colour to stop at point xy and the next colour to start at xy then it would have to include a go to xy without extrusion without extrusion in the gcode.

Prusa-Support, 2024-02-27:
Thanks for reporting. This is supposedly addressed on the slicer' side with the release of PrusaSlicer version [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2).
The Candidate Release has already been published so the final release should be close.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-02-27:
Please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3106.

Michele Moramarco
Prusa Research

closed 2024-02-27 by Prusa-Support

SimonTil, 2024-03-21:
Unfortunately, this doesn't solve the issue for me.
The nozzle still goes back to its last position after M600, squirting a little unwanted second color blob there, and then moves to the first point of the second color line.

I came up with a temporary solution: adding a small wipe tower next to the print, and put the following in Tool Change Gcode:
```
G0 X{wipe_tower_x} Y{wipe_tower_y} F{travel_speed}
M600
```
This gcode-snippet moves the nozzle to the corner of the wipe tower, and squirts the little blob over there. See video below of the gcode viewer running over the script:

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/43646872/bbf9e695-7a8b-49ce-81f4-cfe08b481ac6

You have to understand that this still is a workaround, and not a solution for the existing problem.