prusa3d/Prusa-Firmware-Buddy #3538: [BUG] I had a crash when uploading new wifi name

Niflheim123, 2023-11-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 5.1.0

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

USB drive

**Describe the bug**
had a bsod crash 

**How to reproduce**
For me this happened when I uploaded a wifi config file, took it to the printer, then uploaded the new config to the printer and tried to connect it to the wifi, then I realized that the wifi was with the wrong name (I was on the 5GHz one, the wifi has 5Ghz in the name). Took out the stick, erased from the config file - wifi name - the "5Ghz" characters, put the stick back in and reupload to the printer. 

**Expected behavior**
 New connection to be made to the wifi, Instead it crashed
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13497839/dump_buddy.zip)

Niflheim123, 2023-11-30:
..... removed crash dump, will change credentials

danopernis, 2024-06-05:
Thanks for reporting. This is a known issue, fix is scheduled for 6.1.0 version of the firmware.

closed 2024-11-05 by github-actions[bot]