prusa3d/Prusa-Firmware-Buddy #3539: [BUG] <Enter comprehensive title>

geekcouplettv, 2023-11-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.1.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Start print from Pendrive]

**USB drive or USB/Octoprint**
 Via USB - Drive

**Describe the bug**
 When I want to print gcode or bgcode with manual mmu with 2 or more colors, with 10 or more instances. When i select from the LCD screen to start print it turns black and is not responsive. I must press the reset button or powercycle to access the printer again.

**How to reproduce**
  Make Manual change colors on 1 layer with 2 or more colors. with 10-20 instances of item

**Expected behavior**

**G-code**
  

**Crash dump file**

**Video**
.
[print.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13499319/print.zip)

leejoow, 2023-11-29:
I also have this problem, with only one instance and two colors (total of three tool changes)

geekcouplettv, 2023-11-29:
I am printing Mainly with Multi Material manual still awating for MMU3 to be sended. But Now I have to send many files to do prints. And it's little problematic sometimes.

leejoow, 2023-11-29:
last week I downgraded the firmware to 5.0.1 and it worked again

jonnieZG, 2023-12-03:
Same here. Sample file included.
[mmm-test.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13538628/mmm-test.zip)

TheMoen, 2023-12-24:
I had this issue to, it apears to be something with the wipe tower.
When i disable wipe tower it prints fine.

LucekDev, 2024-01-26:
Printer type - [MK4]
Printer firmware version - [5.1.2]
Original or Custom firmware - [Original]

I observed the same behaviour - gcode was send by PrusaLink, print was started from the printer. First time I pressed the reset without long waiting. Second time I saw that bed is heated (and at the top of the black screen there was eacht time small text ~"Start print"), so I waited longer before pressing reset button. After second try **WiFi credentials were lost**.

Workaround to upload file using PrusaLink with checked "Start print after transfer" worked for me each time - I printed in this way 3 times (once one item, twice 8 items).