prusa3d/Prusa-Firmware-Buddy #3541: [BUG] XL 2T Crash/BSOD at the end of the print, Message: WATCHDOG RESET task:IDLE?

dav991, 2023-11-29:
**Printer type** - XL with 2 toolheads

**Printer firmware version** - 5.1.0
**Original or Custom firmware** - Original
**Optional upgrades** - None

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
Had printed multiple parts today, the printer was never turned off until the issue, this happened once.
After one of the prints, the toolhead was parked at the end, but the printer crashed.

**How to reproduce**
 Only happened once, no patterned identified yet.

**Expected behavior**
No crash should happen.

**G-code**
[06_X2_3-SLOT_MOBO_BRACKET_LONG_V3_BLK_0.6n_0.32mm_PETG_XL_1h34m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13502342/06_X2_3-SLOT_MOBO_BRACKET_LONG_V3_BLK_0.6n_0.32mm_PETG_XL_1h34m.zip)

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13502345/dump_buddy.zip)

**Video**
N/A

**Image**
[image]

mixer3d, 2023-11-29:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3528

markopo1982, 2023-12-06:
Yes, I also have the problem with firmware 5.1.0 with the prusa mini. This happens to me when I print and the watchdog bluescreen appears https://www.youtube.com/shorts/UcE7f9Lbp0A

bkerler, 2023-12-07:
@dav991 Do you have anything connected to the usb port such as a beaglecam or octoprint ? might be related to issue #3554 as well then

Zappes, 2023-12-07:
That sounds interesting. In my case, a Beagle Camera V2 was attached when the problem occurred. In the forum, there's another user who seems to have similar issues - and there's somebody from Mintion participating in the discussion: https://forum.prusa3d.com/forum/user-mods-enclosures-nozzles/beagle-camera-v2-and-mk4/paged/3/#post-688146

bkerler, 2023-12-08:
Small update: it seems it's a buffer overflow and it should be fixed soon.

TolleyLikesRice, 2024-02-27:
Just to add to the pile had the same error after cancelling a print. It cancelled, moved the head out the way, screen flashed to "Loading bootloader" and BSOD'd. Prusa connect reported print status as "Unknown". XL 2T, FW 5.1.2 from the factory. Nothing in the USB port.

hillsandales, 2024-02-29:
I had this happen a couple days ago. My XL 2T had been sitting idle for a week or two. I noticed loud noises coming from it. Upon inspection, the fans were running fast, and had a blue screen WATCHDOG RESET task: IDLE? error. See attached.

[image]

The USB thumb drive had been unplugged from the screen for at least a few days before this happened.  I do have OctoPrint connected to the USB-C port on the back of the XL for monitoring temperatures, but the XL had been idle for awhile when this happened. The Pi 4B also runs a local webcam. I sent the dumb_buddy.bin file to the reports@prusa email address.

dav991, 2024-06-05:
I am already using firmware 6.0.1. I stopped using the included usb drive because it was unreliable months ago.  The issue is not replicable, I only saw it once. I am more than happy to close the issue if the bug is not relevant to the current firmware.

closed 2024-08-12 by github-actions[bot]