prusa3d/Prusa-Firmware-Buddy #3542: [BUG] Prusa Link Beta -  Vorschau Projekte

BigBen2003, 2023-11-29:
**Printer type**
[MK4]

**Printer firmware version**
5.1.0

**Original or Custom firmware**
Original

**Optional upgrades**
Keine Optionale Upgrades

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
Nachdem ein GCode erstellt wurde (Prusa Slicer 2.7.0) wird dieser im Prusa Link in der Vorschau mit einem "?" angezeigt.:

[image]
[image]

Auf dem Display des MK4 wird Vorschau fehlerfrei angezeigt.

[Smartphone Halterung_0.4n_0.2mm_PETG_MK4IS_5h9m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13505740/Smartphone.Halterung_0.4n_0.2mm_PETG_MK4IS_5h9m.zip)

**How to reproduce**
Erstelle ein neues Projekt im PrusaSlicer (PrusaSlicer-2.7.0+win64-202311231457) und exportiere den GCode in eine Datei.

**Expected behavior**
Nach dem eine neue GCode-Datei  hochgeladen wurde, habe ich erwartet, dass der Inhalt der GCode Datei auf der Liste angezeigt wird. Das der Inhalt als Vorschau angezeigt wird                                                                                                                                                                                                                                                      

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
 Es gibt keine Crash-Pump-Datei.
[Smartphone Halterung_0.4n_0.2mm_PETG_MK4IS_5h9m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13505740/Smartphone.Halterung_0.4n_0.2mm_PETG_MK4IS_5h9m.zip)

m-cas, 2023-11-30:
happens on 5.0.1 too (I have not upgraded to 5.1 yet). Could it be a problem of the 2.7 slicer?

BigBen2003, 2023-12-01:
> Could it be a problem of the 2.7 slicer?

Es ist möglich. Bei einem anderen kleinen Projekt traten keine Fehler auf.

BigBen2003, 2023-12-01:
Es liegt auch am Prusa Slicer in der Version PrusaSlicer-2.7.0+win64-202311231457.

Wenn das gleiche Projekt als binärer gcode kodiert wird, treten keine Fehler auf. In der Liste wird die korrekte Vorschau angezeigt.

Wenn allerdings das gleiche Projekt als Ascii gcode kodiert wird, zeigt Prusa Link keine Vorschau an. In der Vorschau wird nur ein "?" angezeigt.

closed 2024-06-10 by danopernis