prusa3d/Prusa-Firmware-Buddy #3546: [BUG] <Enter comprehensive title>

Jam-ehz, 2023-11-30:
**Printer type** - [MINI]

**Printer firmware version** - Original 5.1.0 Final (Nov 23)

Upgrades: Dual-Drive Extruder w/ Filament Sensor

USB Drive and Octoprint, same results. 

Sliced with PrusaSlicer 2.7.0, Input Shaper with the default gcode settings. 0.20mm Structural Settings. 3MF is included in the zip below.

**Describe the bug**
Trying a ringing test - no top or bottom layers, no infill, 1 perimeter, I guess kinda vase mode.
[image]
[image]
[image]

The first 1.5mm are fine, the seam is at the front left corner of the bed. Once the seam switches to the inside corner because of the embossed lettering, it will start to retract, deretract, then snap forward and back again really quickly, leaving about 2.5mm of filament retracted before it starts printing another layer. This retraction leads to underextrusion and only 25% into the layer will the filament catch up and print properly.
This doesn't happen every layer, but most of them. It starts to subside between 9-10mm Z height then starts again.

I've looked at pretty every line of gcode at layer change, and only see the 2.5mm retract/deretract as prescribed in the settings. 

In the videos you can hear the quick snapping and the second video shows the filament left out in between layers.

I've noticed some underextrusion in my prints lately, and this may be related to it..

**How to reproduce**
Printed this 6 times, some reprinted with the same gcode, some resliced and getting the same result.  First 5 with Octoprint, the last time I copied it to USB and get the same result. 

**Expected behavior**
Retraction of 2.5mm and Deretraction of the same between layers.

**G-code and 3MF file**
[gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13517030/gcode.zip)

**Video**
[Link](https://youtu.be/P7-bzMUcAcA)

I held the filament still to get a better look at how much filament is left out:
[Link2](https://youtu.be/Vd0yk81N3EU)

wgentine, 2023-12-01:
The seams on MK4 also changed completely leaving a gap on petg and overlapping on PLA... what a mess.

terminal-cs, 2023-12-02:
same issue on brand new mini+ aswell

Giorgio991, 2023-12-06:
I also have a similar problem with the layer color change on mini +, in addition to the normal retraction, it creates a second one which, once moved into the object, makes a detraction and creates a blob of filament, it also happens with old slicing profiles, it depends on the firmware 5.1.0 final

volwec, 2023-12-10:
I have a similar problem, a few layers are printed correctly and the next ones are with gaps.

terminal-cs, 2023-12-11:
I hope prusa sees this, a few prints have had poor quality surface from seams missing material and streaks with too little fillament from too much detraction from linear advance. I haven't seen any commits related to this or any official replies here

terminal-cs, 2023-12-21:
If anyone at prusa actually reads these issues, it would be nice to have some kind of confirmation that the issue is at least being looked into. Just bought my new mini+ and I'm having this issue, and I'm just kinda sitting here waiting and not knowing if this is ever going to be fixed or not. A lot of the parts I make have to be good quality, and with the holes in them, makes them look broken, or low quality. I would appreciate it if somebody on the development team could give some kind of estimate when it could possibly be fixed. Not even exact, just a general time estimate.

sirgrej, 2023-12-31:
I can confirm this issue with prusaslicer 2.7.1 and mk4 firmware 5.1.2. 

I have made a print with input shaper and the seam looks terrible. So far from my researches it supposed to happen obly  with IS profile but I haven't tested it with the default profile yet

terminal-cs, 2024-01-02:
I'll add that with a 0.6mm nozzle and the 0.6mm nozzle preset, it seems to be less prominent, though is definitely still there a bit. Much more acceptable though

[image]

danopernis, 2024-06-05:
Have you guys tried the latest 6.0.0 firmware and latest slicer? The 5.x.y series of the software probably won't be receiving further patches, but if the issue is still present, we should definitely look into it.

Jam-ehz, 2024-06-05:
I've upgraded to 6.0 and it's been fine with everything I've printed so far, though I haven't tried printing that exact model again. I've just moved up to a MK4, so I won't be able to test it myself.

geekifier, 2024-12-29:
Thanks for submitting this. I have continued experiencing severe print problems ever since the original IS firmware release. The issue occurs when there are many retractions within a small area, and as you have noted, the retraction settings are clearly ignored. I set the retraction distance to .5mm, and yet I see the filament shoot out of extruder, and the horrible grinding sound. I have replaced the hotend, upgraded the extruder, adjusted temperatures, flow, speeds, retraction length and speeds etc.

Everything possible to calibrate has been calibrated. The problem occurs with both 0.4 and 0.6 nozzles, though it is more severe on 0.4. Previously I thought it had to do with small/detailed parts, but now I see it may be related to the seam issues that Mini users have been experiencing.

I can reproduce this every single time with the attached model, sliced with Prusa Slicer 2.9.0, stock Mini IS .4 nozzle preset, using PETG.
[MiniIS-Failed-Print.zip](https://github.com/user-attachments/files/18270956/MiniIS-Failed-Print.zip)

Each one of the vertical elements has a seam, and the failure happens even when reducing the speeds for small perimeters to 10 mm/s! The only thing that changes is the violent slamming of the filament into the extruder input (and the noise) is reduced at slower speeds.

[image]

This is with retraction dialed down to minimum (still ignored by FW) and speed reduced.

[image]

This is with stock settings, usually the filament will end up being broken if it's PLA, in this case it was PETG. It looked like a clogged nozzle, but I was able to extrude after stopping the print.

https://github.com/user-attachments/assets/387eee0e-fc42-49f2-b636-d0a93fa96c12

geekifier, 2024-12-29:
Not long after my last post, it seems I have finally manage to print the model!

Two key changes:

**Seam position**: Rear (previously aligned)
**External Perimeters First**: True (previously False).

It printed without major issues (other than some minor stringing, most likely from previously reduced retract distance)!

It's a Christmas Miracle! :). So there seems to be some issue between the firmware and the slicer settings when it comes to seams/many small details printed close together, on the Mini.

This was using "full" Prusa Mini IS 0.4 speed settings, whereas previously I could not get it to work even at 10 mm/s!

[image]

GCODE Preview, note the seam position.

[image]

blackpoplar, 2025-05-05:
Seeing similar issues with mini firmware 6.2.2 PrusaSlicer 2.9.2. Using input shaper printer definition and "0.20mm structural" settings results in unpredictable but repeated excessive retraction after the first few layers. This results in print failure. Prints correctly using legacy (no input shaper) printer definition and "0.20 mm quality" settings  with no excessive retraction.  See pictures. using Prusament PETG filament and stock filament definitions.

[image]
[image]