prusa3d/Prusa-Firmware-Buddy #3549: [BUG] Setting static IP causes prusa connect to always refuse connection

markaabo, 2023-12-02:
**Printer type** - MK4 Kit

**Printer firmware version** - 5.1.0

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**
  When setting static IP via prusa_printer_settings.ini (As mentioned in #3101) prusa connect always refuses the connection never allowing the printer to get a code and makes it ALWAYS fail acquiring of code.
  
 Additional info: my printer is connected via WIFI (ethernet not available) and when trying to set the static ip via wifi router settings, the printer never connects and the router generates an error causing the router to block internet. which is why i have set the static ip via the ini file, the reason why i need to set a static ip is stuff like Home assistant and networking issues, which don't like the ip of the printer to change. and the reason why the ip changed i don't know, literally one day the printer just wanted a different ip even tho the normal ip was available for use.

**How to reproduce**
  Use prusa_printer_settings.ini (As mentioned in #3101) to set static IP
  Try to acquire code for prusa connect

**Expected behavior**
  For prusa connect to work and allow the connection

TojikCZ, 2024-02-15:
Hi, unfortunately setting static IP on buddy is clunky. I have managed to do it. It CAN work.

You have to set the DNS servers in the [network] section of the config

so you have your wifi or eth ::ipv4 section with ip, mask and gw + the network section with dns.

To load the settings, go into Settings, System, Load setting from file, the other load wifi settings or connect settings read just a part of the file, not the whole thing.

ChiefPoints, 2024-04-18:
This is still an issue for static IP's, I'll edit my current prusa_printers_setting.ini to add in the DNS section - I improperly assumed it would pick it up from the gateway address. Prusalink works fine, just couldn't get it to register without Prusa Connect. Perhaps the devs can update the error message it gives with "hostname not found", instead of just saying "Failed"

ChiefPoints, 2024-06-27:
Yes, this issue should not be closed until it's actually fixed.

danopernis, 2024-06-27:
Hey folks, we tried to reproduce this on 6.0.3 and according to our testing, this is working. We used following INI file for testing:

```
[network]
hostname = knoblet
dns4=10.67.22.1

[wifi::ipv4]
type=static
addr=10.67.22.88
mask=255.255.255.0
gw=10.67.22.1
```

FWIW, we are working on a GUI to improve debugging network-related problems, because at the moment printer doesn't really tell you what is wrong with the configuration. I can't give you ETA on that. In the meantime, please double check that all of those field are setup correctly. You can load the INI file via Settings -> System -> Load Settings From File

Prusa-Support, 2024-09-10:
Hi everyone. Because the connection and static IPs can fail in many different ways, and for several reasons that are not strictly related to the printer's firmware, in FW [6.1.2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.1.2) our developers implemented the new "Network Status screen" in the hope to help advanced users debugging connection at various levels.

The new interface provides information about latency and rate of data package delivery failures at several levels of the connection: `connect` (our server), `gateway` (router), `DNS server` (local/provider server).

> [!TIP]
> If DHCP works but static doesn't, it may quite likely be the case of a problem of static IP configuration, or a problem of settings on the router' side.
In case of large networks, the chosen IP address should be outside of the DHCP range to avoid conflict of address.

In addition, users now can set their own hostname and forget about IPs via the new "User-configurable hostname" feature implemented in the same firmware release.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-11-22:
I hope you have found the provided information useful.
Now, please allow me to close this issue as not strictly related to firmware.
Also, our firmware has been enhanced with hostname editing.

There is probably not much that we can do about the router and the local network but I hope setting a custom _Hostname_ will greatly simplify the access to Prusa Link. The reference article has been updated including everything the user needs to know about the use of hostnames on up-to-date Prusa printers.
https://help.prusa3d.com/article/how-to-set-a-static-ip-address_695648

Michele Moramarco
Prusa Research

closed 2024-11-22 by Prusa-Support