prusa3d/Prusa-Firmware-Buddy #3550: [FEATURE REQUEST] Improve time offset to include GMT+13

Kingston802, 2023-12-02:
**Printer type** - MK4

**Is your feature request related to an existing problem? Please describe.**
Currently, New Zealand is in the GMT+13 timezone, but the maximum offset on my MK4 is +12 meaning I can't have an accurate time on my printer, would be great if the offset could account for this edge case

**Describe the expected functionality**
  Time Zone Offset allows for values up to 13

tracernz, 2024-01-16:
This is the first and only real issue I encountered after assembling my new Mk4 kit. Nice to see it’s already reported 👍.

Prusa-Support, 2024-04-04:
This issue is solved by FW [6.0.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC1), which is a Release Candidate and therefore considered to be a rather stable pre-release.

Feel free to give it a go and share further feedback.

Michele Moramarco
Prusa Research

closed 2024-04-04 by Prusa-Support