prusa3d/Prusa-Firmware-Buddy #3551: [BUG] Octoprint not connecting to MINI on FW5.1

ps-igel, 2023-12-02:
**Printer type** - MINI

**Printer firmware version** - 5.1

**Original or Custom firmware** - Original

**Describe the bug**
 
Octoprint [1.9.3] is not connecting to the MINI running FW 5.1 via Serial connection.

Octoprint just reports:

```
octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
octoprint.util.comm - INFO - Connecting to port /dev/ttyACM0, baudrate 115200
octoprint.util.comm - INFO - Failed to connect: Port /dev/ttyACM0 is busy or does not exist
```

**How to reproduce**

Update MINI to 5.1, try to connect via Octoprint.

**Expected behavior**

Octoprint should correctly connect.

Reverting to MINI FW 4.3.4 fixes the connection issue instantly.

Please let me know if I can provide any more log output.

One additional remark. I am running Octoprint in a docker container from a Synology NAS. The docker container will not fire up even with FW 4.3.4 if the printer is not powered on. Only if it is powered on the container will startup.

However with FW 5.1 the container will not start running regardless if the printer is powered on or not. Hopefully this can be helpful to analyse the main problem. Maybe there is a fundamental problem with serial connection functionality in 5.1.

stonesbg, 2023-12-29:
Just upgraded to Printer FW 5.1.2 using octoprint 1.9.3 and same issue serial port won't connect.

Octoprint is running off a raspberry pi in my case

faunris, 2024-02-18:
Same issue, but with MK4. Works excellently on MacOS, but on a raspberry pi doesn't work.

ps-igel, 2024-05-03:
I now updated to Prusa Firmware version 6.0.0 and octoprint 1.9.3 but still no luck. If I switch back to firmware 4.3.4 it works again.

I also tried settings baudrate in Octoprint to AUTO, doesn't work

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyACM0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Offline"
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyACM0@115200, /dev/ttyACM0@250000, /dev/ttyACM0@230400, /dev/ttyACM0@57600, /dev/ttyACM0@38400, /dev/ttyACM0@19200, /dev/ttyACM0@9600
Trying port /dev/ttyACM0, baudrate 115200
Connecting to port /dev/ttyACM0, baudrate 115200
Could not open port /dev/ttyACM0, baudrate 115200, skipping
Trying port /dev/ttyACM0, baudrate 250000
Connecting to port /dev/ttyACM0, baudrate 250000
Could not open port /dev/ttyACM0, baudrate 250000, skipping
Trying port /dev/ttyACM0, baudrate 230400
Connecting to port /dev/ttyACM0, baudrate 230400
Could not open port /dev/ttyACM0, baudrate 230400, skipping
Trying port /dev/ttyACM0, baudrate 57600
Connecting to port /dev/ttyACM0, baudrate 57600
Could not open port /dev/ttyACM0, baudrate 57600, skipping
Trying port /dev/ttyACM0, baudrate 38400
Connecting to port /dev/ttyACM0, baudrate 38400
Could not open port /dev/ttyACM0, baudrate 38400, skipping
Trying port /dev/ttyACM0, baudrate 19200
Connecting to port /dev/ttyACM0, baudrate 19200
Could not open port /dev/ttyACM0, baudrate 19200, skipping
Trying port /dev/ttyACM0, baudrate 9600
Connecting to port /dev/ttyACM0, baudrate 9600
Could not open port /dev/ttyACM0, baudrate 9600, skipping
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
```

ps-igel, 2024-05-04:
And I noticed something else while plugging the printer in via USB. It is connected to a Synology NAS. When I call `dmesg` I see different logs between when I plugin the server with printer fw version 4.3.4 and 6.0.0.

When connecting with fw 6.0.0 I get only get the message 

`usb 1-1: new full-speed USB device number 12 using xhci_hcd`

but when connecting after flashing fw 4.3.4 it shows

```
[13321002.042185] usb 1-1: new full-speed USB device number 13 using xhci_hcd
[13321002.759149] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[13321002.767385] usbcore: registered new interface driver cdc_acm
[13321002.775375] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

[image]

Did something change in the way the printer handles USB connections?

ps-igel, 2024-05-06:
I got it sorted out. I followed [this tutorial](https://mariushosting.com/synology-how-to-add-usb-support-on-dsm-7/) to get the USB to be correctly identified as ttyACM06 again.

closed 2024-05-06 by ps-igel