prusa3d/Prusa-Firmware-Buddy #3552: [BUG] XL does not turn off hot end after unload

morphias2004, 2023-12-03:
**Printer type** - XL 5 tool head

**Printer firmware version** - 5.1.0 production

**Original or Custom firmware** - Original

**Describe the bug**
Unloading filament from 1 or multiple tool heads leave the hot ends running at the unload temp after docking.

Have to manually tell the tool heads to cool down.

**How to reproduce**
Unload filament from 1 or more tool heads.

**Expected behavior**
Turn off hot end after docking.

[image]

DoubleStrike, 2023-12-04:
What would be better is to add a timeout value such that if the printer is not printing, and a tool is docked and heated up, it auto-shuts off the tool after XXX seconds.  Say, 15 seconds as an example.  Can't really see a downside here.  Keeping a nozzle heated and active for no reason wastes power and is a potential safety risk.

BrienKing, 2023-12-10:
> What would be better is to add a timeout value such that if the printer is not printing, and a tool is docked and heated up, it auto-shuts off the tool after XXX seconds. Say, 15 seconds as an [...]

It does that already.  The printers have a safety timer in them that shuts off the heaters with no activity.  I believe it's 30 minutes.

morphias2004, 2023-12-10:
> It does that already. The printers have a safety timer in them that shuts off the heaters with no activity. I believe it's 30 minutes.

Maybe so, but 30mins is way too long.

The heating should be turned off as soon as (or within 5-10secs) the unload or load is completed. It should not be left in the tool changer or docked for 15-30mins at full temp. It's just cooking whatever filament is in the hotend or causing oozing.

Jetro2203, 2023-12-11:
This "problem" also exists on the MK4, it would make sense to me if the hot end stays hot when I selected the "change filament" option. But I would prefer that the hot end immediately cools down once the filament is unloaded if I selected the "unload filament" option.

TechnoSwiss, 2024-01-22:
Agree, this one keeps tripping me up and it's kind of annoying to have to remember to go into the temperature meeting to turn off the hot end. Waiting for the 30 min. timeout it definitely too long, have enough oozing problems with the printer, leaving the filament fully loaded with the hot end still at full temp after a filament change just exacerbates it.

morphias2004, 2024-01-22:
This is still an issue 5.1.2.

It also sets Tool 1 to 170C if you load or unload filament in other tools.

And they get left on for 30mins, not 15mins as someone else stated. That's not safe.

morphias2004, 2024-04-24:
Still there in production release of 6.0.0

morphias2004, 2024-06-27:
This is still an issue that really needs to be addressed. It's not a major change to the FW.

danopernis, 2024-06-27:
Not sure if this is a bug or feature, so I am letting our product department to handle that...

Dorito203, 2024-09-07:
I have the same problem (firmware 6.1.2+7894). 
No matter in which extruder I change the filament, E1 is heated to 170°C afterwards. This remains set even if you start a print with a different extruder. I have noticed that E1 is still heating up after hours of printing.

danopernis, 2024-09-17:
Pasting potentially useful comments from #4107 

> For example, if I change the filament on tool 4, tool 1 is always heated to 170c even though I haven't changed anything... I then have to deactivate it manually.

> I'm seeing this too, but I think it's a bit more general than this.
> After changing/loading/unloading the filament on any tool other than T1, T1 is set to the standby temperature (170C), while the active tool remains at the change temperature.
> 
> Expected behavior: the active tool is set to the standby temperature, all other tools remain unchanged

DoubleStrike, 2024-10-06:
That last comment from #4017 is critically important as it is a DIFFERENT bug than what is stated above.  There are 2 distinct issues:

1.  Loading filament into tools 2-5 ALSO preheats Tool 1 and leaves it heated for no reason. This needs to be fixed ASAP, as **this is a definite safety issue** and could lead to burns from an unexpected heated nozzle.
2. After a filament load, the related tools should be set to cooldown **no more than 15-30 seconds** after the operation ends.  Not doing so, as has been stated several times above and on forums, is a safety issue and also can actively damage filament sitting in the hotend.  The 30 minute safety timeout is **way too long** for this scenario and is only a last-ditch emergency timeout.

morphias2004, 2024-10-06:
That this is still open almost a year later is a poor reflection on how @prusa3d  view safety issues.

There seems to be more focus is adding 'bright and shiny' rather than fixing things that never should have made it to production firmware.

DoubleStrike, 2025-02-01:
Any thoughts on this?  We are sitting at the start of 2025 and there are two potential safety issues here, as referenced in my post above.  These really need to be addressed as bugs, as they have very real safety implications.

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3552#issuecomment-2395266002

DoubleStrike, 2025-11-24:
> Starting from FW 6.2.0, unused tools are actually shut down at the beginning of the print, once the user is done with all filament operations and ready to print (or power down). https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0#:~:text=now-,turn,-off
> 
> On top of this, during auto-load filament operations, the user is theoretically "_still_" required to select the single relevant tool for auto-load to work, as highlighted in the most recent release notes - nothing happens to inactive tools. https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha#:~:text=relevant%20tool
> 
> I seem to understand that the current firmware [A] automatically disables unused tools at the beginning of the print or [B] automatically disables "forgotten heaters" at the end of a safety timer. Please let us know if there are other or more or less common scenarios that are not covered by the current heater deactivation rules (including start of the print and safety timers).
> 
> When it comes to [safety timers](https://help.prusa3d.com/article/safety-and-safety-timer-messages-core-one-xl-mk4-s-mk3-9-s-mk3-5-s_857246) and customization, I understand that some users may have personal preferences, and I'm not sure if the current values are ruled by specific safety standard regulations (it may be the case). However, heater timers may be customized via `M86` - _not stored after restart_. _Requests that are closely related to safety timers though, should ideally be posted as a new issue with a clear title and a rather specific description._
> 
> Michele Moramarco Prusa Research

Great feedback, but I think you've missed a use case here.

If one is loading filament in a couple of tools, but is NOT starting a print, the tools sit there, heated, until the 30 minute safety timer elapses.  The point I am making is that the safety timer here is far too long, and is not obvious to the average user. 

After a filament is loaded in Tool A, and I load a filament in Tool B, I would not expect Tool A to remain heated. This makes little sense, and further, keeping it hot for up to half an hour could be very bad in a scenario where someone comes back to the printer to clean Tool A and finds it scorching hot. This also has the unfortunate side effect of continuing to subject the filament in tool A to higher temperatures for an extended period of time, which is normally undesirable.

While the "turn off unused tools at print start" is important, I don't think it addresses the core issue here at all. The point was that after switching to Tool B, Tool A should be *immediately* cooled - **a safety timer should not even be in the picture**.

I can open a new issue tagged as such, but was trying to avoid more tracking work for you guys.

morphias2004, 2026-02-04:
That this case, like a number of others I and other users have lodged, is still open and outstanding is just plain insane.

This has a very simple fix in firmware. Prusa continue to ignore the customer base and continue to treat the XL like a bastard red-headed step-child.

Instead, we see developer focus on useless 'features' that look good in blogs and social media posts instead of addressing core functionality issues.