prusa3d/Prusa-Firmware-Buddy #3554: [BUG] BSOD on first layer print

bkerler, 2023-12-03:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.1.0 stable]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint**
   USB flash drive

**Describe the bug**
  Blue screen of death while printing the first layer

**How to reproduce**
  Open the file and print (Sent via prusaslicer 2.7.0 as bgcode)

**Expected behavior**
  Regular print with no BSOD.

**G-code**
[print_file.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13538669/print_file.zip)

**Crash dump file**
[dump_buddy2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13538675/dump_buddy2.zip)

**Video**
  No video.

**Picture of BSOD**
[image]

mixer3d, 2023-12-03:
@bkerler https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3528

bkerler, 2023-12-04:
@mixer3d I removed the details before uploading, and yes, I also am concerned about this, but this has nothing to do with my serious bug.

mixer3d, 2023-12-04:
ok, better you removed the file, there were your credentials there before, pattern is the same for every printer...

bkerler, 2023-12-04:
@mixer3d. No, I had critical stuff removed before. But yes, network details were there as well and the api key (but that can be changed). Cleaned that up now as well. Thx for pointing that out. @Prusa-Support should really take care to remove sensitive stuff from the dump file, not funny.

bkerler, 2023-12-07:
Update: The crash also appears when the printer is just in idle, not printing anything as soon as the /dev/tty1 is connected (doesn't have to be beagle V2). This issue didn't appear with the 5.0 firmware.

For each crash, the LR and PC is identical and stack as well.

bkerler, 2023-12-07:
The PC/LR itself seems to point to the crash_dump::trigger_crash_dump() which isn't really helpful :(

bkerler, 2023-12-07:
**So here's the test setup:**
1. Turn on the printer
2. USB-C connected to the back port of the beagle cam, beagle cam powered externally
3. Power on the beagle cam
4. Start the beaglecam app, select the printer

Printer was started at 12:32, BSOD crash happens in MK4 being idle at 12:40.

bkerler, 2023-12-07:
@vorner @dragomirecky @Derevin @LukasLendvorsky could anyone of you please provide assistance what to do next ?

bkerler, 2023-12-07:
The bug is unrelated to the beaglecam btw. ... I was able to get the same crash using just a regular pc serial port connection (took about 15 minutes). But with beaglecam the crash is easily reproducable.

Tomcus, 2023-12-07:
Hi, We are currently working on fixing a similar issue. We are currently trying to reproduce this issue (from your instructions, thanks for those), since we didn't have a reproducer so far. If we are able to reproduce it, we will try your suggested solution. This way you don't have to void your warranty.

bkerler, 2023-12-07:
@Tomcus Thanks a lot. If there is anything else I can do to support you (more logs, anything) please let me know. The Beaglecam v2 seems to trigger it very efficiently with a window of around 8 minutes. Doing it manually using a python script, it's hard to reproduce but sometimes it also crashes after a longer period.

Tomcus, 2023-12-07:
@bkerler If you can provide (gist, pastebin, ...) the python reproduction script, it would be great.

Tomcus, 2023-12-08:
@bkerler We were able to reproduce the issue. So you don't have to provide the script :wink:. Bu thank you anyway for the reproducer. It helped a lot.

bkerler, 2023-12-08:
That's great 👍 What was the issue ?

bkerler, 2023-12-08:
Sounds pretty much like that's the issue. If you want you can send me a signed debug/beta firmware to test in order to confirm

bkerler, 2023-12-17:
Seem the issue is still there with fw 5.1.2

Tomcus, 2023-12-18:
@bkerler Yes, the 5.1.X was locked before we started working on this issue. Currently there are some regression issues that were made during the changes, that also need to be fixed before publishing. So hopefully everything will be ready before 5.2 releases. At worse it will be part of any future release.

serhiihoroshko, 2023-12-19:
This issue was reproduced on 5.1.2 Firmware for Original Prusa MK4
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13718012/dump_buddy.zip)

**Precondition:**

- Raspberry Pi 4 Model B (Connection via USB Type-C)
- OctoPrint 1.9.3
- OctoPi* 1.0.0 (build 2023.10.09.151442)
- Python 3.9.2

**Steps to reproduce:**

1. Connect to the OctoPrint
2. Click the "Power" icon on OctoPrint website
3. Select the "Reboot System" option

topchiyev, 2023-12-25:
Hi. I have the same problem on Prusa Mini+ 5.1.0 with Beagle camera V1.

idaadi, 2024-01-11:
I have the same issue.  does anyone know of a lower firmware which doesnt do this?

Mawos, 2024-01-23:
I'm having the same issue on my Mk4 and Mini with firmware 5.1.2. Looking forward to the resolution!

CruisnGrrl, 2024-02-17:
Also having this issue with my Mk4 with an odroid c4 running armbian + octoprint connected using a USB A-C cable.

TLMcNulty, 2024-02-20:
@CruisnGrrl same, but I'm on 5.1.3. Have you tested 5.2.0?

bkerler, 2024-02-25:
5.2.0 isn't released yet for the MK4, but the 5.2.0 MK4 release should have the needed usb fixes applied.

N3orun, 2024-03-08:
same here with my raspi and octoprint. Has there been a release date set for the new firmware?

midnightcruze, 2024-03-23:
Same issue here!

Prusa-Support, 2024-04-09:
Hi everyone, there will be no FW 5.2 for MK4 but FW 6.0.

The next firmware release is available as Release Candidate [6.0.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC1), which is considered to be a rather stable pre-release.

It should include bug fixes merged in the last two months, so feel free to give it a go and to share further feedback.

Michele Moramarco
Prusa Research

bkerler, 2024-04-19:
Bug was fixed in 6.0. Thx !

closed 2024-04-19 by bkerler