prusa3d/Prusa-Firmware-Buddy #3558: [BUG] missing many thumbnails on MK4, which also won't print

gfixler, 2023-12-04:
printer: Prusa MK4 (assembled by Prusa)
printer firmware version: 5.1.0 (original)
PrusaSlicer version: 2.7.0+linux-x64-GTK3
tweaks/upgrades: none, all stock
data transfer: USB thumbdrive

**The bug**
After upgrading to 5.1.0, from 5.0.1, new prints, using .bgcode, wouldn't show thumbnails on the MK4's screen. The files showed 3 lines of info about print time, etc., but the actual values were all dashes.

I then noticed a lot of the old prints on the drive - a few that were recent .bgcode, and many that were older (but still recent - only had this printer a month) - had the same problem. Testing a half-dozen, when trying to print, the red light on the heat bed plug comes on, and the screen says PRINT at the top left, and shows the USB stick in the top right, and that's it. The rest of the screen is black. It just sits there until maybe 15-20 minutes later, when it beeps and tells me on the screen that it's stopped due to inactivity. I could start some prints with thumbnails (stopped them before they actually started printing, but they weren't stuck at the black screen, and seemed normal). I can't exit. Turning the knob does nothing. Pressing it beeps, but does not else. I have to hit the reset button to reboot to get out.

My slicer changes were minimal, things like Extruders: 2, Single Extruder Multi Material: on, Enable ironing: on, and 2 modified PLA entries (just to lower print temps a bit for my filament). I also tried wiping out all my changes and going with defaults, including a single extruder.

**Temporary fix**
The only thing that fixed it was rebooting the machine with 5.0.1 (not 5.1.0) on the USB stick, pressing the button twice at bootup, and flashing the old one back again.

**How to reproduce**
Upgrade to 5.1.0 and look through both .gcode and .bgcode prints. Some number of files, exported from recent and the current version of PrusaSlicer (see top for version) may be missing thumbnails, and will not begin printing. 

Here's an example of a simple STL I couldn't get to print, nor show a thumbnail, failing as detailed above:
https://www.printables.com/model/96736-fillet-gauge/remixes

**Expected behavior**
I expect to see thumbnails on the MK4 screen, and to be able to print the files.

**G-code**
[FilletGaugeFlat_WHEELER_0.4n_0.2mm_PLA_MK4IS_14m.gcode.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13550267/FilletGaugeFlat_WHEELER_0.4n_0.2mm_PLA_MK4IS_14m.gcode.txt)

tkircher, 2023-12-05:
Duplicate of #3510

tmhedesign, 2023-12-05:
Same setup here, probs with Thumbs. 

Also top layer  left corner is printing bad, almost like nozzle would go too low. tested on 2 mk4s running stable 5.1 firm

gfixler, 2023-12-07:
> Duplicate of #3510

That issue is about thumbnails only. I also have the problem that if there isn't a thumbnail, the machine is stuck, and I can't print. Is that issue expected to manifest along with the thumbnail issue?

gfixler, 2023-12-15:
Things seem to have evened out for me naturally after moving up to 5.1.0 again, so I'm going to close this issue.

closed 2023-12-15 by gfixler