prusa3d/Prusa-Firmware-Buddy #3559: [ENHANCEMENT] Prusa Link Password edit/remove

sydios, 2023-12-06:
Hello,

it would be great if you could change the password yourself or disable it completely. 
Not all printers are in an open network and need protection. It's a bit annoying especially because it's done with an .htaccess kind of protection. On mobile, it blocks the whole browser if you don't enter the correct password.

Gompje, 2023-12-11:
Second this. 

its annoying as hell. especially given it's done with a .htaccess file like you said which eliminates any browser saving or password manager. Please let us decide if we need this or not. 

I don't care if it's on by default, but for the love of God let us remove it.

tkircher, 2023-12-15:
This is part of why most of us want Octoprint support. Even if Prusa had an army of devs working on their custom interface, there's only so much they can do with the hardware. I'm pulling the built-in wifi module as soon as I can.

CZDanol, 2024-06-05:
Internal ticket: BFW-5639

CZDanol, 2024-06-05:
Note: Completely disabling the password is not doable, it would make the printer way more vulnerable to attacks.

Btw, you can apparently use http://user:password@ip/ to connect if you want to skip the obnoxious auth dialog.

CZDanol, 2024-06-05:
Also note: There's a quite nice integration of PrusaLink with HomeAssistant :)

danopernis, 2024-06-05:
Also note: user:password@ip doesn't work because of "secure" `window.fetch` https://stackoverflow.com/questions/75539407/why-fetch-doesnt-allow-credentials-in-the-url

See also https://github.com/prusa3d/Prusa-Firmware-Buddy-Private/pull/5067

danopernis, 2024-06-05:
> Btw, you can apparently use http://user:password@ip/ to connect if you want to skip the obnoxious auth dialog.

This doesn't work because `window.fetch()` is "secure" (see [stack overflow](https://stackoverflow.com/questions/75539407/why-fetch-doesnt-allow-credentials-in-the-url)) but maybe could work if PrusaLink extracted the fields from URL and sent them using headers.

The-Wurst, 2024-06-05:
> Note: Completely disabling the password is not doable, it would make the printer way more vulnerable to attacks.

I don't mean that it's disabled by default but that every user can choose to do so if they want. I have a secure network and don't need the PW. Or at least make the PW editable so you can choose your own. Forced security is always annoying, especially on a pro machine like this.

CZDanol, 2024-06-05:
> I don't mean that it's disabled by default but that every user can choose to do so if they want. 

I understand that.

> I have a secure network and don't need the PW.

Secure network is not enough. Without the PW, people could theoretically hack the printer even through a javascript in a website ad.

CZDanol, 2024-06-05:
I understand the annoyance, I hate it as well :/ But we just cannot do something this unsafe.

The-Wurst, 2024-06-05:
> I understand the annoyance, I hate it as well :/ But we just cannot do something this unsafe.

Then why not add an option to choose your own password? Or switch away from the .htaccess so you can use a password manager. The solution right now is just not usable and super annoying. But again, why force your sense of security on the owner? You can add a disclaimer so that if you disable the PW, it's at your own risk. The machine is almost $3000 let me choose my settings.

CZDanol, 2024-06-05:
> Then why not add an option to choose your own password?

Working on that ;)

lubarb, 2024-08-01:
> Or switch away from the .htaccess so you can use a password manager. 

With 1Password on chrome, when the login popup is present, I can open 1password and drag the "maker" username and then the password to the fields in the login prompt. A pain, but a better workaround for me than having to type them in.

emente, 2024-11-13:
> Secure network is not enough. Without the password, people could theoretically hack the printer even through a javascript in a website ad.

The are numerous web standards that protect against this. 

What about reading a a file from USB storage that can override username and password?

vojty, 2024-12-17:
I got sick of this so I created a simple proxy server that drops the authentication (it automatically injects the auth headers with the credentials stored in a file). I run it on my old Raspberry Pi 2. In case anybody is interested https://github.com/vojty/prusa-link-proxy Use with caution as it's basically a backdoor. I tested it with my Prusa Mini

tkircher, 2024-12-17:
It might also be worth looking at octoprint, now that Prusa has gotten support to about 70%. You can easily configure it to auto login, no need for a username or password.

e-basteln, 2025-04-16:
Thanks for the additional cross-reference. But are we going to get user-definable usernames and passwords anytime soon?

I find the status quo especially annoying when I open PrusaLink from within PrusaSlicer (on the "Physical Printer" tab). Every time I first connect to the printer, it asks me for the impossible-to-remember password. Apparently there is no possibility to have it auto-filled? 

Yet the password is plainly readable a few clicks away, in PrusaSlicer's Printers > Edit physical printer tab. So there is no security provided by the password query, but it makes me jump through a few hoops to go over to the Printers tab, open the Edit dialog, copy the password, go back to the PrusaLink tab, past the password. Not a great experience.

lewisd32, 2025-04-18:
I would like to suggest another good reason to allow purchasers of Prusa printers to set a password: Buddy board replacements.
I've had to have my buddy board replaced under warranty, and being unable to set the Prusalink password on the new board to the same thing it was on the old board is causing me great headaches.

I have many physical printers configured in PrusaLink, with the same IP and password, because of needing different profiles for different nozzles, as well as a separate MMU3 profile. Changing the passwords on all of these is frustrating. Changing them back again if I switch back to the old buddy board (if it turns out there's nothing actually wrong with it) will be doubly frustrating.

I use the HomeAssistant integration, and it doesn't allow for editing existing instances of devices. I cannot change the password there. I need to delete the printer integration and reconfigure it, which will break all the automations and scripts that reference it.

This could be easily avoided if the printer allowed me to set the password on it, like virtually every other device does.

shuhaowu, 2025-09-15:
lol so I made a reverse proxy that strips the digest auth, allowing for you to access the prusa link server _without authentication_. There's even a prebuilt docker image:

```
docker run \
  -e DIGEST_AUTH_REMOVAL_PROXY_USERNAME=<username> \
  -e DIGEST_AUTH_REMOVAL_PROXY_PASSWORD=<password> \
  -e DIGEST_AUTH_REMOVAL_PROXY_BACKEND=http://prusalink-coreone \
  -p 8080:8080 \
  ghcr.io/shuhaowu/digest-auth-removal-proxy:v1
```

Now if you combine this with nginx and auth_request, then there's a very nice solution. I'm currently running this in production.

airtonix, 2025-10-06:
> Note: Completely disabling the password is not doable, it would make the printer way more vulnerable to attacks.
> 

@CZDanol but it's not on HTTPS... so what's the point?

airtonix, 2025-10-15:
I can't help but feel that things are the way they are because Prusa is required by contract or law to implement these changes in order to secure income through private contracts.

They have a "Malicious Compliance: Do the bare minimum in order to satisfy the poorly written regulations or contact terms" vibe about it.

tkircher, 2025-10-15:
> I can't help but feel that things are the way they are because Prusa is required by contract or law to implement these changes in order to secure income through private contracts.

 European laws probably do require terrible security practices, but they could have a subsidiary in some other country with a copy of the signing key releasing firmware for everyone else, where you have to pinky swear that you aren't in the EU. I am willing to take Prusa's official responses at face value, though, meaning that they are just incompetent.