prusa3d/Prusa-Firmware-Buddy #3562: [BUG] <Enter comprehensive title>

markopo1982, 2023-12-06:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** -prusa mini

**Printer firmware version** - 5.1.0

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor / Revo Hotend

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive 

**Describe the bug**
  I found an error, my USB stick doesn't work properly with the firmware 5.1.0, there's always a blue screen, the printing is always aborted, but when I print with Begglecam everything works fine

**How to reproduce**
The printing stops with the USB stick

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  yes there is a dump file

**Video**
https://www.youtube.com/shorts/UcE7f9Lbp0A

bkerler, 2023-12-07:
The issue might come from the beaglecam vs usb serial connection. See issue #3554. Can you try if your issue appears as well if the beaglecam is completely disconnected from the printer ?

danopernis, 2024-06-05:
No, this has nothing to do with beagle camera, it is working as good as any serial line printing would. It is a problem with USB stick which becomes irresponsive. @markopo1982 new versions of the firmware are trying to better recover from USB stick failures, can you try that?

closed 2024-08-12 by github-actions[bot]