prusa3d/Prusa-Firmware-Buddy #3563: [BUG] Firmware update for release version of 5.1.0 corrupted files on flash drive

Scrazzled, 2023-12-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.1.0 (release , not alpha)

**Original or Custom firmware** - [Original]

**USB drive**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
Updating the firmware from 5.0.0 to 5.1.0 corrupted all the files and directories that existed on my flashdrive, and it created 133 new items which were all also corrupt files/directories. 

After copying the new firmware file to the flash drive and updating the MK4, 133 new items (files and directories) were created and are corrupted. Most of the files are just under 1Gb in size, some  > 1.8 Gb 
Windows cannot open them. Examples of filenames: 
layer_ch.ANG
type:ext.ern
F1500 M.204
54 Y108..999

Examples of directories created (I added the double quotes so show that the period  was part of the directory name):
"869 g1 x.80." 
"75 G1 X8.1.3"
**How to reproduce**
  Please describe steps to reproduce the behavior.

I had a directory for my gcode files, which was corrupted and I could no longer access it. I unfortunately did not back up this flash drive and I ended up losing dozens of files. 

**Steps**
Backup and remove the existing firmware file from the flashdrive. 
Copy the firmware file for 5.1.0 to the flashdrive
Connect the flashdrive to the MK4 and press the reset button
Update the firmware
update the wifi firmware
After wifi was finished updating and was connected, and UI was returned to the main menu, I clicked the print button, then the screen locked up as it loaded 136 folders and files, some of which were almost 2Gb in size. 
I removed the flashdrive from the MK4, and connected it to my PC and examined the contents of the drive. Everything except the authors text file was corrupted. 

**Expected behavior**
Upgrading firmware doesn't corrupt all my files.

Workaround, reformatted flashdrive copied the new firmware file again to the flashdrive.

JacobChrist, 2024-05-16:
Not sure if it's related but I uploaded a .gcode file with a really long file name to V5.0.0+12061 and the Flash drive got corrupted and now looks something like this:

[image]

JacobChrist, 2024-05-16:
Actually, my issue seems more closely related to: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2422

Prusa-Support, 2024-06-14:
Thanks for reporting.

Same as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2422#issuecomment-2168053764, this problem is potentially related to other forms of data corruption and interferences reported in several ways, but I would keep it open because it describes data corruption from a rather specific perspective.

Please consider sharing a cloned image of the corrupted flash drive via email at reports@prusa3d.com, together with a short description of the problem and of the steps to possibly reproduce the problem.

Michele Moramarco
Prusa Research

closed 2024-08-22 by github-actions[bot]