prusa3d/Prusa-Firmware-Buddy #3567: [BUG] MINI FW 5.1 Cancel single object not available on display

yaronze, 2023-12-09:
Hi,
I am working with MINI Prusa printer firmware 5.1.0 and Prusa Slicer 2.7.0. According to firmware release notes the option of Cancel Single Object via printers panel is a new feature.
I tested it with my MK4 printer and it works great. I tried to do same with MINI printer and i cant find the "cancel single object" menu item while printing. 
I even tried many options like create instances or sequential printing - still not possible to cancel single object in the MINI.
Am i missing something? or it is a bug in firmware?

mix579, 2023-12-09:
Release notes say 
Cancel Object (XL, MK4/MK3.9)
so not supported for Mini (at this point). I assume not enough memory to hold the object maps.

yaronze, 2023-12-10:
Thanks. I missed this notice in the document.

bkerler, 2023-12-10:
I assume this issue can be closed as currently not supported and thus no bug ?

yaronze, 2023-12-11:
yes

closed 2023-12-11 by yaronze