prusa3d/Prusa-Firmware-Buddy #3569: [BUG] <Crash when changeing filament for other color during print>

Prusamk4, 2023-12-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - 
MK4

**Printer firmware version** - 
5.1

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
None

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint
USB Flash drive

**Describe the bug**
  When i want to change the color, it tells to get out the filament, and then it chrashes, nothing more goes on.

**How to reproduce**
  Simply try to change the filament, usunug the option filament change

**Expected behavior**
  Get out the filament when it tells to do so, then it happens

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Prusamk4, 2023-12-12:
I made a print right now. When it comes to change the filament, as you can see, it tells to please remove the filament. I do so, but it goes no further. 
May be it's crasht, or in a endless loop? Pressing any buttoms does not work at all, power off or reset works.

[image]
[image]

bkerler, 2023-12-12:
Did you test if the filament sensor is working ? Or did you recalibrate the filament sensor ?

Prusamk4, 2023-12-12:
Hello

 

Now it seems to work.

 

Maybe I used a too old filament. After a glance to manufacturing data, it was 2012…. Too much dirt, I think.

 

Unmounting and clean the sensor it started to work again.

 

Many thanks for the advice.

 

Eduard

bkerler, 2023-12-15:
Great :) Can you then close this issue as resolved ?

closed 2023-12-15 by Prusamk4