prusa3d/Prusa-Firmware-Buddy #3571: [BUG] Fuzzy Skin Regression in Firmware v5.1.0 (Regardless of Input Shaping State)

dstulken, 2023-12-10:
**Printer type** - Prusa Mini+ and others (see below)

**Printer firmware version** - v5.1.0 as compared to v4.4.1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** -  USB

**Describe the bug**
The printing behavior of parts with fuzzy skin appears to have substantially regressed with Mini firmware v5.1.0. 
Under the new firmware, the ability to reproduce small lateral offset detail appears to have been lost - instead, surfaces are essentially flat with a random distribution of large zits.  What used to be able to create nice organic silky textures instead now outputs what looks like sandpaper.

This bug does not appear to be a PrusaSlicer issue.  The slicing renders in PrusaSlicer v2.7.0 appear as expected, showing the fine detail that the Mini can no longer reproduce.   Files sliced with prior versions of PrusaSlicer, from before the introduction of Input Shaping, which previously printed fine on the Mini in firmwares v4.4.1 and before, exhibit the issue under firmware v5.1.0.  So this appears to be a change in the way the Mini interprets the gcode.   

Three additional data points:
- Files sliced using identical PrusaSlicer settings and printed on alternate printers (such as my Ender 6 running Klipper) produce the correct output texture, matching that of the Mini on firmware v4.4.1 and earlier.
- A file re-sliced on PrusaSlicer v2.7.0 using the new InputShaping profile, also produces the same incorrect print output on a v5.1.0 firmware Mini.
- We updated a second of our Mini+ units, and it shows identical before/after behavior, which rules out a coincidental hardware issue.

**How to reproduce**
- Place a generally flat-sided item on the build plate.
- Set layer height to 0.2mm
- Set material to PETG
- Set Fuzzy Skin to outside walls
- Set Fuzzy Skin Thickness to 0.05mm
- Set Fuzzy Skin Point Distance to 0.2mm
- Observe the texture as rendered in the slice preview
- Print the part on a Mini with FW version 5.1.0, and observe that the output doesn't match the render preview
- Print the part on a Mini with FW version 4.4.1 and observe that the output matches the render preview

**Expected behavior**
The output of the fuzzy skin, for prints with input shaping not enabled, should match the PrusaSlicer preview (which is how it prints on Mini firmware v4.4.1 and before, and also how it prints on non-Prusa printers).

**Attachments**
Three images are attached.  It is difficult to convey a glossy texture in a 2D photograph, but hopefully you can see the difference.

The first image is how the texture appears in the slicer.
[image]

The second image is how the texture appears in printed parts.  
On the left is a print from Mini firmware v4.4.1, and on the right, the same print on firmware v5.1.0.
[image]

The third image shows a pair of parts printed in PLA, both using the exact same gcode file.  
On the left is a print from firmware v4.4.1, on the right is a print from v5.1.0.
[image]

dstulken, 2024-03-15:
(Bumping so this doesn't get flagged as stale.)

dstulken, 2024-04-16:
(Monthly "not stale" bump.)

danopernis, 2024-06-27:
@dstulken can you please test this with latest firmware?