prusa3d/Prusa-Firmware-Buddy #3572: [BUG] 5.1.0 Watchdog Idle

Mushbeeguy, 2023-12-10:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [e.g. 4.0.5, ...]

**Original or Custom firmware** - [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Mushbeeguy, 2023-12-10:
Printer type - [MK4]

Printer firmware version - [5.1.0]

Original or Custom firmware - [Original]

Optional upgrades - [none]

USB drive or USB/Octoprint
USB drive

Describe the bug
Occasionally BSOD Watchdog ide error requiring reset. Normally happened when idle. Now hapened 4hrs into an 11hr print
How to reproduce - Just wait. 
Please describe steps to reproduce the behavior.
 Leave printer powered up.

Expected behavior
Not to crash while sitting or at least continue printing.
G-code
Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

Crash dump file
Attached dump_buddy.bin As dump_buddy.DMP
[dump_buddy.dmp](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13626508/dump_buddy.dmp)

Video
JPG attached. Note progress on print.
[image]

 This is a factory assembled printer . A second on site printer does not have this problem...yet.

bkerler, 2023-12-10:
Do you have any camera / serial port attached to the printer ? If yes, then issue #3554 might be responsible.

Mushbeeguy, 2023-12-10:
I did read that and it may very well be my ineptitude.

closed 2024-06-27 by github-actions[bot]