prusa3d/Prusa-Firmware-Buddy #3575: [FEATURE REQUEST] Cancel object using PrusaConnect web UI

B-0-B-B-Y, 2023-12-10:
Since https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3073 has been merged in, and stable releases of FW 5.1.0 and PrusaSlicer 2.7.0 now fully support this functionality via the physical printer menu, it would be nice to extend this ability to the PrusaConnect web UI, as being physically present on-site with your printer isn't always possible. For scenarios where one part of an entire batch job has failed, it would be nice to be able to remotely still save the entire print process and just cancel the failed part/object via the web UI.

**Describe the expected functionality**
Menu option in the PrusaConnect web UI exists to allow canceling an object, able to select which object to be canceled and sending that command to the printer, remotely, without the need to physically interact with the printer.

BugsBunny1403, 2024-03-18:
I support your request, and confirm the interest in adding this functionality. I will remain attentive to the rest of the events.

Kosaro, 2024-04-06:
This already exists, but the option to do so only appears sometimes. I'm not sure why.

wafflecart, 2024-04-06:
> This already exists, but the option to do so only appears sometimes. I'm not sure why.

Did not know cancel object existed in prusa connect! Will see if I can see it next print 😎

tjstansell, 2024-04-19:
I printed a 16 object print yesterday after updating to the 6.0.0 firmware (MK4) and using prusaslicer 2.7.4 and I couldn't find any way to access the cancel object feature in prusa connect.  I did initiate the print using prusalink, so not sure if that changes anything.  I would hope not.  🤷

wafflecart, 2024-04-21:
You access it by clicking on the print on the list then on next screen scroll down to see the build plate. You can click on models there to cancel.

whoisdecoy, 2024-04-23:
> You access it by clicking on the print on the list then on next screen scroll down to see the build plate. You can click on models there to cancel.

I was able to see this on my MK4. Will this be available on XL with official 6.0.0? My XL is on RC3

[image]

rawlk, 2024-04-23:
> I was able to see this on my MK4. Will this be available on XL with official 6.0.0? My XL is on RC3
> 

It showed up for me on my not-yet-6.0 XL today. 

Thanks! A huge help, now just have to figure out how to tell it to resume after canceling the problematic object remotely.

danopernis, 2024-05-17:
@B-0-B-B-Y can this issue be closed now, or is there anything else?

rawlk, 2024-05-17:
It seems to not be consistent. I tried several times in the last few days to find this feature when printing with my MK4 MMU and couldn't ever cancel object from prusa connect.

ChrisHeerschap, 2024-07-01:
I've been sending to connect via a physical printer with the PrusaConnect API in 2.7.x, that didn't work, and now I'm printing direct to PC from 2.8.0 and it's still not showing the cancel object option on anything. Printer always shows cancel object and it works, but somehow the cancel object never shows up in PrusaConnect.

B-0-B-B-Y, 2024-08-05:
> @B-0-B-B-Y can this issue be closed now, or is there anything else?

It seems the behaviour is inconsistent, so would be good to keep this open still until we can get confirmation that it works on a consistent basis

closed 2024-10-15 by github-actions[bot]