prusa3d/Prusa-Firmware-Buddy #3576: [ENHANCEMENT] Make English only version available

im-evo, 2023-12-11:
**Printer type** - [MINI]
**Optional upgrades** - NA

**Is your suggested improvement related to an existing problem? Please describe.**
Hello, please add a build target for English-only firmware. At this time, it's only possible to install multi-language ones. Thanks!

**Describe the expected functionality**
An option to install English-only firmware.

bkerler, 2023-12-17:
That's already possible by setting the additional cmake option "TRANSLATIONS_ENABLED" to "NO".

im-evo, 2023-12-18:
> That's already possible by setting the additional cmake option "TRANSLATIONS_ENABLED" to "NO".

Nice, then the only thing missing is a build for releases so that a user is not forced to build it themselves or have a translation they will never use. 

It can also help track language preferences overall by the number of downloads.

loglow, 2024-01-11:
Agreed. It is odd and confusing to have to pick one of six non-English languages to install. I actually came here because I was slightly confused about it.

digiltd, 2024-01-11:
@loglow i agree, even though each file has the word `english` in it, when looking at them stacked in a folder list my brain just sees the differences i.e. the second country (stupid brain)

[image]