prusa3d/Prusa-Firmware-Buddy #3577: [BUG] MK4 Selftest Heating Silicon sock

sphxdr, 2023-12-11:
**Printer type** - MK4 

**Printer firmware version** - 5.1.0 (and older 5.0.1)

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** Not relevant 

**Describe the bug**

On the Prusa MK4 (Kit) it gives a Error when trying the selftest (Nozzle heater check) with a Silicon Sock(https://www.prusa3d.com/de/produkt/prusa-nextruder-silicone-sock-set/). I tried both with the Setting on and off and it keeps failing on Firmware 5.0.1 and 5.1.0. Without sock (and setting deactivated) it completes the test successfully!

**How to reproduce**
Start Selftest (Nozzle heater check) with a Silicon Sock and the Setting active. 

**Expected behavior**
Selftest should succeed!

bkerler, 2023-12-11:
Nozzle heater test does pass fine with prusa silicon sock attached on my MK4, just tested with 5.1.0 stable. Maybe your thermistor is defective ?

sphxdr, 2023-12-12:
Hi!

My test is failing only when the Silicon sock is on the heat block. 

If I do a test without it it passes (always that the hardware setting is deactivated as it should be)

bkerler, 2023-12-12:
Interesting. For me it works fine with sock settings being turned on under hardware option and with sock on the nozzle.
And without sock and sock setting turned off it works as well. What max temperature do you read when the heater is tested before cooldown (with sock) ?

gamefreak613, 2024-01-14:
I'm also seeing this behavior.

fecsketalp, 2024-01-16:
Having the same issue.

sphxdr, 2024-01-24:
Hi @bkerler, it got Up to 168-171ºC I tested today the new Firmware 5.1.3 and sadly this is still happening.

bkerler, 2024-01-24:
Yes, expected to be fixed in 5.2 probably.

fecsketalp, 2024-01-24:
> Having the same issue.

In the meantime it turned out, that the heater cartridge is faulty. The self test was failing without socks after a while.

bkerler, 2024-01-24:
@sphxdr Mine goes to 200°C with 5.1.3 as expected and passes the test with sock installed. Sounds pretty much like your heater cartridge or thermistor is broken, contact support. Except you might haven't got the prusa 0.4 nozzle installed, that might also cause issues for the selftest I assume.

sphxdr, 2024-01-24:
😅I tried again with my 0.4 Brass Nozzle (I previously left my ObXidian 0.6 from last print) and it went up to about 180°C. If something is wrong I would guess it's the heater block as I already changed the thermistor. I must still insist that it is a little strange that it passes every test without sock. Maybe the test with a sock has tighter tolerances…

gamefreak613, 2024-06-06:
Still an issue!

closed 2024-08-13 by github-actions[bot]