prusa3d/Prusa-Firmware-Buddy #3578: [BUG] Violent collision displaces steel sheet, has apparent z offset error, risks printer damage (XL)

tg73, 2023-12-11:
**Printer type** - XL multi-tool

**Printer firmware version** - 5.1.0 release

**PrusaSlicer version** - 2.7.0 release

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint**
Prusa-supplied USB key (shipped November 2023) initiated via PrusaConnect, using bgcode.

**Describe the bug**

This is about printing this model (https://www.printables.com/model/610843-enxlosure-minimalist-and-printable-prusa-xl-enclos) on an XL, particularly the supplied per-oriented "Plate2". Several people have had problems printing this particular plate of parts, so this seems to be a reproducible problem under a variety of circumstances. See the comments here: https://www.printables.com/model/610843-enxlosure-minimalist-and-printable-prusa-xl-enclos/comments

At 97% completion on this print I heard a loud bang from the printer and went to investigate. There had been a collision that had enough force to dislodge the print sheet from the print bed. The photos show the position of the print sheet after the collision. All the parts remained fully attached to the print sheet. Notably, the part which was hit was fully attached and not warped. The tall part of that part was not warped (see photos with ruler alongside). The print sheet was not warped.

The top of the part that was hit appears to have a nozzle-shaped indentation inside the top layer of the part. It gives the impression that there was a z-lift move where z height was not actually increased at the start of the move, but did decrease at the end of the move, plunging the nozzle into the print - then the printer continued to move, dragging the part and print sheet with it.

I have since removed the nozzle in question for inspection and it does not have visible damage, but it might have non-visible damage. It does now require using some force with the spanner to remove and reinsert it into the heatblock, so the thread might have been damaged. It seems like a miracle that the nozzle/heatbreak was not bent given the amount of force involved in the collision. The print sheet was twisted out of location and jammed up against the left z-axes threaded rod. There is slight marking to the edge of the print sheet.

I went to the trouble of writing a program to check the gcode to see if the fault was there, but it does not appear to be - at least when checking the gcode as converted by PS from the bgcode. But please don't take my word for it. Anyhow, this is why I consider it to be a firmware issue.

I was using well-dried eSUN PETG with the stock Generic PETG profile, XL 5T Input Shaper 0.32 SPEED 0.6mm nozzle. Ambient temp around 21C, no drafts. The last layer printed was at about 225mm (by later measuring the cold print).

I have reported my experience to Prusa support and am awaiting a response.

<details>

<summary>Photos (click to expand)</summary>

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

</details>

**How to reproduce**
Print the affected plate of parts.

**Expected behavior**
A successful print.

**G-code**
[3mf and bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13636763/3mf.and.bgcode.zip)

kll, 2023-12-11:
I had the same issue on the same parts but not using the pre-oriented plate. I created my own plate with just the two large parts and using PLA for supports. I was in an adjacent room when the initial collision happened and the noise got my attention so I went to investigate. I saw another collision so then I grabbed my phone to capture it on video. At the time the video was taken it had already had at least 2 violent collisions so the tops of the parts are already messed up. I don't know if the collision I captured is exactly the same as the original, but I don't see how a little bit of loose filament on top of the part could cause a collision this bad so I assume what I captured is the z offset error causing it to dig the nozzle into the part or catch a significant portion of the nozzle on the side of the part.

[video](https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/16741/26a9e5ca-df0d-4861-9545-8fc41424ab1d)
[enxlosure_bottom_back_both_0.6n_0.25mm_PETG,PLA_XLIS_9h52m.bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13637692/enxlosure_bottom_back_both_0.6n_0.25mm_PETG.PLA_XLIS_9h52m.bgcode.zip)

JeremiahGillis, 2023-12-11:
I had the same issue when printing this twice. Here is my original print with the pre-oriented plate 2:

[image]

[image]

I reprinted the two large pieces while remaining by the machine during the end. The nozzle started catching so I ended up supporting them with my hands while it finished. I could see the nozzle catching while on the same piece, not just when moving from piece to piece during the print. Here is a video of the nozzle catching:

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/13179985/758ecaa9-0e48-45fa-85ef-25cb675c888d

The model EnXLosure_Bottom_BR seems to have the issue more than EnXLosure_Bottom_BL. I wonder if the overhang is creating enough of a lip for the nozzle to catch on.

downtm, 2023-12-11:
Same issue here on my 5T XL.  I wish I took photos, not sure why I didn't.

I was using the pre-made plate 2, and also heard a loud bang from the other room.  I came to find my unit was telling me that I needed to manually re-seat my toolhead, which involved taking it off the head manually and re-seating it in the dock.

By that point, after starting the print again it was too late as the collision already had shifted the part off the build plate and wasn't able to resume where it crashed.  There was a small indentation in the side of EnXLosure_Bottom_BR where the nozzle hit.

Later on, when printing another set of parts I re-calibrated my toolheads as I swapped from a .4 to .6 nozzle.  It was at this point that I discovered the prior failed print had hit my Tool head 4 dock so hard it put it out of alignment.  I have since re-aligned the dock and tightened the screw, but it was very concerning that the toolhead hit the dock with such force while printing.

Is there a bug with the z-ramp hop or w/e that moves the toolhead back from the print?  I am assuming there is something at the top that is causing it to shift further back that it is supposed to, exceeding the normal boundaries of the print area.  I wasn't using a priming tower or anything for this, but concerning nonetheless.

I would also note that PrusaSlicer said the print job was fine and there were no out of bounds indicators.

tg73, 2023-12-13:
> It was at this point that I discovered the prior failed print had hit my Tool head 4 dock so hard it put it out of alignment. I have since re-aligned the dock and tightened the screw, but it was very concerning that the toolhead hit the dock with such force while printing.

@downtm Could you inspect the rear of the tool that was active when it collided with parked tool 4 to see if there was any visible damage? You might see marks from the main connecting "spikes", or where the dwarf cover hit the extruder latch (see image below). You can see [here](https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-general-discussion-announcements-and-releases/psa-prusa-connect-control-may-move-without-homing-and-collide-with-docked-tools/) what happened when the toolchanger on my printer was rammed into the parked tools when there was no active tool.

Here are some contact point examples (posed by hand when the machine was powered off):

[image]
[image]
[image]
[image]

downtm, 2023-12-20:
> @downtm Could you inspect the rear of the tool that was active when it collided with parked tool 4 to see if there was any visible damage? You might see marks from the main connecting "spikes", or [...]

I inspected all of my tools and do not see visible damage on the back of them.  I think my toolhead 4 dock being slightly loose from vibration, though not ideal at all, saved it when it shifted slightly back during the collision.

tg73, 2024-06-06:
Waiting for Prusa to respond.

Prusa-Support, 2024-09-06:
Thanks for bringing this to our attention.

Layer shifting, unfortunately, as usual, may result in occasional collisions as observed on other printer models.
In the specific case of Prusa XL with XY-core though, this may result in erratic crashes all around the frame which may result in misalignments of the tools and of the locking bars.
In other words, this may not necessarily be the case of a firmware bug.

Usually, layer shifting may be caused by collisions onto a lifted spot of the print, loose belts, loose parts of the tool-head, and other hardware/assembly problems which may not find solutions at the firmware level.
This goes without saying that there may be very specific episodes that deserve rather specific considerations to be made based on the individual case, through a proper troubleshooting session, with the help of our Customer Support.
https://help.prusa3d.com/article/customer-support_2287

In this specific case, we have a very specific print file and more than one report based on the same experience, printing this very specific print project!
Again, this may not seem to be the case of anything wrong with the firmware so you may not expect to find a definitive solution in this repository, which ultimately concerns the firmware. However, as long as the problem is reproducible on multiple printers "in optimal conditions", this may be an interesting case study to improve printability in similar scenarios. 

At first glance, we are looking at a tall poorly supported, thin, tall, and slated element, where the short layer-time prevents optimal cooling performances. I may speculate that layer after layer, the element becomes rather wobbly, and the fact that the layer has very limited time to set down makes the plastic smeared and sticky, hence keener to move along with the nozzle.
This is nothing but speculation, but this is where my first thought rather naturally leads to.

Perhaps a wise use of advanced and experimental settings in Prusa Slicer can improve layer time or reduce collisions.
Perhaps this is easily reproducible due to the print being so tricky, but not reproducible on all printers in optimal conditions.
This may leave room for testing scenarios but so far, I see no direct correlation with firmware problems. 

Michele Moramarco
Prusa Research

tg73, 2024-09-08:
@Prusa-Support (Michele) - as regards how this issue is related to firmware: this is where I think Prusa has the opportunity to innovate. You've had to disable TMC-based (stepper controller) collision detection due to the requirements of other features such as input shaping and phase stepping - fair enough. But the XL is loaded with sensors, so there are other firmware-based ways to detect collisions such as using the strain gauge or accelerometer(s) (could also be the accel's in the parked tools, not just the active tool). Detecting at least severe collisions - be it with a print, or the machine colliding with itself - would be great, and would reduce the chances of damage to the machine through the collision and wasted material due to prints continuing despite a layer shift.

Prusa-Support, 2024-09-24:
If this is your ultimate request, you may want to make it clear in the object and the description of the issue, keeping it as concise as possible. This issue really looks more like a troubleshooting request, not suitable for this GitHub repository but to be discussed individually with Customer Support.

I'd like to assure you though that our product managers and developers have already considered these options without any doubt, and actually, many more potentially viable alternatives ("viable" theoretically only, so far) to improve the collision detection algorithms and any other safety and recovery options.
In fact, such improvements even occurred for several years all over the production and development timeline of the previous series (MK2, MK3, MK2.5 ...). All of this time we proved to care about product strengthening much more than "sell and replace" market rules and we will pursue this goal.

The information on this issue has been collected as general feedback but there is probably not much we can draw to the advantage of the firmware that strictly concerns this GitHub repository. I suggest closing this issue and working on the problem together with our Customer Support.

Michele Moramarco
Prusa Resaearch