prusa3d/Prusa-Firmware-Buddy #3579: [BUG] <can't compile MK4 firmware>

kgolger, 2023-12-12:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.1.0]

**Original or Custom firmware** - [Original]

**Describe the bug**
can't compile MK4 firmware 5.1.0

tryed to compile as user pi:

git clone https://github.com/prusa3d/Prusa-Firmware-Buddy.git - ok
mv Prusa-Firmware-Buddy TEMP/ - ok
cd TEMP/Prusa-Firmware-Buddy - ok
python utils/build.py --generate-bbf - error!

Switching to Buddy's virtual environment.
You can disable this by setting the BUDDY_NO_VIRTUALENV=1 env. variable.
Traceback (most recent call last):
  File "/home/pi/TEMP/Prusa-Firmware-Buddy/utils/build.py", line 832, in <module>
    main()
  File "/home/pi/TEMP/Prusa-Firmware-Buddy/utils/build.py", line 623, in main
    bootstrap.switch_to_venv_if_nedded()
  File "/home/pi/TEMP/Prusa-Firmware-Buddy/utils/bootstrap.py", line 211, in switch_to_venv_if_nedded
    os.execv(str(venv_bin_dir / 'python'),
FileNotFoundError: [Errno 2] No such file or directory

wbr,
Karl

seppelicous, 2023-12-12:
I can confirm this issue. 
I get the same error macOS and windows 12. both with python 3.12.1

bkerler, 2023-12-12:
It's easy to fix. install virtualenv, then do a `python -m venv .venv` in the git directory and it will work just fine using the build script.

seppelicous, 2023-12-12:
I appreciate your quick support!
It works and now it is compiling 😊

seppelicous, 2023-12-12:
Update:
Maybe the reason for the following error has the same root:

`C:\Users\PrickeS\Downloads\prusafirmware>python utils/build.py --preset mk4
Switching to Buddy's virtual environment.
You can disable this by setting the BUDDY_NO_VIRTUALENV=1 env. variable.

C:\Users\PrickeS\Downloads\prusafirmware>Collecting ecdsa~=0.18 (from -r C:\Users\PrickeS\Downloads\prusafirmware\requirements.txt (line 2))
  Using cached ecdsa-0.18.0-py2.py3-none-any.whl (142 kB)
Collecting polib~=1.2 (from -r C:\Users\PrickeS\Downloads\prusafirmware\requirements.txt (line 3))
  Using cached polib-1.2.0-py2.py3-none-any.whl (20 kB)
Collecting pyyaml~=6.0 (from -r C:\Users\PrickeS\Downloads\prusafirmware\requirements.txt (line 4))
  Obtaining dependency information for pyyaml~=6.0 from https://files.pythonhosted.org/packages/2b/9f/fbade56564ad486809c27b322d0f7e6a89c01f6b4fe208402e90d4443a99/PyYAML-6.0.1-cp312-cp312-win_amd64.whl.metadata
  Using cached PyYAML-6.0.1-cp312-cp312-win_amd64.whl.metadata (2.1 kB)
Collecting littlefs-python==0.8 (from -r C:\Users\PrickeS\Downloads\prusafirmware\requirements.txt (line 5))
  Obtaining dependency information for littlefs-python==0.8 from https://files.pythonhosted.org/packages/09/8b/5b2f9b0b29a2bae7aee648259e92e798c259d6e1455ccbc79e92da0842e2/littlefs_python-0.8.0-cp312-cp312-win_amd64.whl.metadata
  Using cached littlefs_python-0.8.0-cp312-cp312-win_amd64.whl.metadata (5.8 kB)
Collecting numpy~=1.24.0 (from -r C:\Users\PrickeS\Downloads\prusafirmware\requirements.txt (line 6))
  Using cached numpy-1.24.4.tar.gz (10.9 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\cli\base_command.py", line 180, in exc_logging_wrapper
    status = run_func(*args)
             ^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\cli\req_command.py", line 248, in wrapper
    return func(self, options, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\commands\install.py", line 377, in run
    requirement_set = resolver.resolve(
                      ^^^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\resolver.py", line 92, in resolve
    result = self._result = resolver.resolve(
                            ^^^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 546, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 397, in resolve
    self._add_to_criteria(self.state.criteria, r, parent=None)
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 173, in _add_to_criteria
    if not criterion.candidates:
           ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_vendor\resolvelib\structs.py", line 156, in __bool__
    return bool(self._sequence)
           ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 155, in __bool__
    return any(self)
           ^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 143, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 47, in _iter_built
    candidate = func()
                ^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\factory.py", line 206, in _make_candidate_from_link
    self._link_candidate_cache[link] = LinkCandidate(
                                       ^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 293, in __init__
    super().__init__(
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 156, in __init__
    self.dist = self._prepare()
                ^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 225, in _prepare
    dist = self._prepare_distribution()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PrickeS\Downloads\prusafirmware\.venv\Lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 304, in _prepare_distribution
    return preparer.prepare_linked_requirement(self._ireq, parallel_builds=True)
[... 20 lines cut ...]
[... cut ...]
  File "<frozen importlib._bootstrap>", line 1325, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 929, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 994, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "C:\Users\PrickeS\AppData\Local\Temp\pip-build-env-0m7etvnk\overlay\Lib\site-packages\setuptools\__init__.py", line 10, in <module>
    import distutils.core
ModuleNotFoundError: No module named 'distutils'`

bkerler, 2023-12-12:
I pushed a PR #3582 to fix this bug.

kgolger, 2023-12-12:
DISTRIB_DESCRIPTION="Ubuntu 22.04.3 LTS"
Python 3.10.12

Thanks @bkerler
Installed virtualenv and did a python -m venv .venv in the git directory.
Following result after send python utils/build.py --genarate-bbf

Switching to Buddy's virtual environment.
You can disable this by setting the BUDDY_NO_VIRTUALENV=1 env. variable.
Building mini_release_boot
Traceback (most recent call last):
  File "/home/pi/TEMP/Prusa-Firmware-Buddy/utils/build.py", line 832, in <module>
    main()
  File "/home/pi/TEMP/Prusa-Firmware-Buddy/utils/build.py", line 800, in main
    result = build(configuration,
  File "/home/pi/TEMP/Prusa-Firmware-Buddy/utils/build.py", line 328, in build
    config_process = subprocess.run([str(get_dependency('cmake'))] + flags,
  File "/usr/lib/python3.10/subprocess.py", line 503, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/pi/TEMP/Prusa-Firmware-Buddy/.dependencies/cmake-3.22.5/bin/cmake'

wbr,
Karl

bkerler, 2023-12-12:
Karl, the exec format error normally only appears if you mix up 64bit vs. 32bit. Also I think you mean "--generate-bbf" ?

Prusa themselves seem use this command line (as an example):
`python3 utils/build.py --preset mk4 --build-type release --bootloader yes --generate-bbf --generate-dfu --no-store-output --version-suffix=-BETA+5729.PR3570.B5 --version-suffix-short=-BETA+5729 -DCUSTOM_COMPILE_OPTIONS:STRING=-Werror -DDEVELOPMENT_ITEMS_ENABLED:BOOL=yes`

kgolger, 2023-12-12:
You are right - of course "--generate-bbf"
I have installed Ubuntu 22.04.03 LTS ARM64 (Pi400).
For this reason, I will probably not be able to use the Prusa build.py. I'll try it on a 32-bit system.
Thanks!
wbr,
Karl

bkerler, 2023-12-12:
It could very be that the problem is arm64 and not 32bit/64bit. I assume the cross-compile tools are only meant to be used with x86/x64. But you could try to emulate a x86/x64 system on the arm64.

kgolger, 2023-12-12:
Thanks!
As i meant, i'll try it later on a x86 system (also Ubuntu 22.04 LTS).
wbr,
Karl

kgolger, 2023-12-12:
compilation now runs on x86 and x64 systems (both tested). 
Closing Bugreport - Thanks to all!
wbr,
Karl

closed 2023-12-12 by kgolger