prusa3d/Prusa-Firmware-Buddy #3580: [BUG] Printer continues moving head after last object is cancelled

kb9ysc, 2023-12-12:
**Printer type** -  MK4

**Printer firmware version** - 5.1.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
Printed via USB drive.  

**Describe the bug**
 Upon cancelling the last (in my case only) object,  the printer continued to move the head around-- it did cancel the object,  but since there were no more objects in the print,  it didn't extrude anything after I cancelled the object.  I guess the bug is in the behavior after I cancel the last object on the plate.  

**How to reproduce**
Start a print of one object.   Then Cancel the object through the tune menu.  (not the stop print button)

**Expected behavior**
Cancelling the last (or only) object should behave similar to the stop button.    So I guess it could be a line of code that says if number of object printing is one,  that instead of running the code to cancel that object, it just runs the stop print code. 

**G-code**
I don't think this is relevant to the report..  I'd think any gcode will do.  (I will try testing with other gcodes to see if the behavior is the same though)

**Crash dump file**
 The printer didn't crash,  so I don't have a crash dump file  

**Video**
I can attach a video if needed.

bkerler, 2023-12-12:
Having a look at the firmware source code, the word "cancel" might be misleading. In fact it's skipping the object in the current print process which means that it doesn't stop the print process, but just skips the object itself. 

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/15d5ae0b317a37b69b8e7a4720711b962c25b57d/lib/Marlin/Marlin/src/feature/cancel_object.cpp#L57

Thus I assume this is expected behaviour and not a bug.

kb9ysc, 2023-12-13:
Very true.  it works with multiple objects.  but if you skip the last object,  it's skipping virtually everything, and the interface becomes rather slow because it's doing a lot of skipping of gcode.    if there are no objects being printed,  it might make more sense to stop/cancel the print.  The behavior makes sense, I guess,  in a way, but I'd suggest that the skipping of the last or only object in the print be considered an attempt to stop the print job completely.   I debated whether 'bug' was the right place to make this comment.

bkerler, 2023-12-13:
Guess it would make sense to change from bug to feature request ;)

closed 2024-06-27 by github-actions[bot]