prusa3d/Prusa-Firmware-Buddy #3584: [BUG] MK4 5.1.0 CRASH on Octoprint reboot

taquitos, 2023-12-12:
**Printer type** - [MK4]

**Printer firmware version** - [5.1.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [N/A]

**USB drive or USB/Octoprint** - [**OctoPrint**]

**Describe the bug**
Rebooting the pi 4b while connected triggers a crash.
OctoPrint 1.9.3

**OctoPrint Plugins installed**
- BGCode (0.2.0)
- Camera Settings (0.4.3)
- OctoPod Plugin (0.3.16)
- UPS (0.0.1)

**How to reproduce**
1. Connect OctoPrint
2. Click "Power" icon on OctoPrint website
3. Select "Reboot System"
4. 💰

**Expected behavior**
I expect a graceful handling of the disconnect

**Crash dump file**
[dump_buddy(x2).zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13651075/dump_buddy.x2.zip)

mixer3d, 2023-12-12:
@taquitos https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3528

taquitos, 2023-12-12:
> @taquitos #3528

Well f-me. Thank you for pointing this out

bkerler, 2023-12-12:
@taquitos most probably related to this issue #3554

taquitos, 2023-12-12:
> @taquitos most probably related to this issue #3554

Yup, the more noise customers make about the issue the more likely the issue will be prioritized, I'm guessing 😅

bdlabitt, 2024-01-30:
Quite reproducible and annoying for me as well.
MK4, 5.1.2, Octoprint 1.9.3, Octopi 1.0.0
Reboot the RPI4 and watch dog time out occurs on the MK4.  Honestly, not acceptable behavior.
[image]

bkerler, 2024-01-31:
This bug is already addressed in issue #3554 and is said to be fixed the 5.2 release

closed 2024-06-27 by github-actions[bot]