prusa3d/Prusa-Firmware-Buddy #3586: [BUG] XL multitool can move active tool into parked tools after a collision causing damage

tg73, 2023-12-12:
**Printer type** - XL multitool

**Printer firmware version** - 5.1.0 release (original)

**Describe the bug**
If collision detection is disabled (which it is by default, and must be disabled for input shaper), the XL can experience collisions leading to significant XY position misalignment, and can then drive the active tool into the parked tools with significant force causing damage.

**How to reproduce**
Any print which can lead to a collision, for example see #3578.

**Expected behavior**
The XL should not be able to move the active tool into the parked tool area except when performing tool change operations and when in a known pre-homed, collision-free state with accurate XY positioning.

**Comments**
The XL now has input shaper (IS) as a release feature (FW 5.1.0, PS 2.7.0) and so Prusa is is actively encouraging and sanctioning its use. However, crash detection must be disabled, and is disabled by default. This means that there is no protection against the active tool being driven with force into the parked tools due to misalignment after a collision. I'd suggest that the XL should either require that collision detection is enabled, or it should have an additional safety sensor that detects when the y carriage moves into the "tool change operations only" zone. This would allow the firmware to detect dangerous movement during normal print operations and then take appropriate action. The additional sensor could be for example a hall effect sensor attached to the chassis with a magnet on one of the y-carriage parts. Even if collision detection is enabled, the additional sensor would provide a second line of defence against the printer damaging itself due to an undetected collision. As far as I am aware, this is the first Prusa printer that has been able to damage itself like this, so a different attitude towards disabling crash detection is needed. For example, the worst that will happen to a MK3 after an undetected collision is hitting an endstop.

tg73, 2023-12-13:
[Here](https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-general-discussion-announcements-and-releases/psa-prusa-connect-control-may-move-without-homing-and-collide-with-docked-tools/) you can see an example of the force involved and the kind of damage that can result (in this example, all tools were parked, so the damage was to the toolchanger itself).

Here are some contact point examples (posed by hand when the machine was powered off):

[image]
[image]
[image]
[image]

tg73, 2024-06-06:
Waiting for Prusa to respond.

danopernis, 2024-09-02:
@tg73 is this actual bug with actual reproducer, or are we just speculating here?

tg73, 2024-09-02:
@danopernis The link in a [previous comment](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3586#issuecomment-1853789366) shows the real-world damage made to the toolchanger head by a dock collision. It therefore seems logical to assume that damage could result to a loaded toolhead. [Here](https://youtu.be/3PobTR-zJOU?si=rexgrKTtPmf8Ktga&t=81) is a video showing the toolhead crash into the dock after a layershift (not my video, just from yt searching). Note the camera shake from the force of the collision.

closed 2024-11-19 by github-actions[bot]