prusa3d/Prusa-Firmware-Buddy #3587: Printer hangs up when trying to print a multi color print

KC5810, 2023-12-13:
### Description of the bug

When trying to start a print using multiple extruders and the option "single extruder multi material" is checked the printer does nothing. I have the file loaded on a USB drive as a bgcode. This process worked perfect before the recent update to the 5.1 printer firmware.

### Project file & How to reproduce

[SwirlBase_0.4n_0.2mm_REDGRN,PETG,PETG_MK4_19m.zip](https://github.com/prusa3d/PrusaSlicer/files/13655190/SwirlBase_0.4n_0.2mm_REDGRN.PETG.PETG_MK4_19m.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 11

### Printer model

MK4

sdvaletone, 2023-12-14:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3543

& 

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3539

Color change prints work just fine it seems, its the use of single extruder MMU that does not work for some reason.

Raveshaw, 2023-12-17:
Hi,

I have the same issue. After starting a Multi Material Print without using a MMU, I just get a Black screen.

If the extruder was empy before, I get the screen back with the Filament insertion Screen, as soon as I feed Filament into the Extruder. But when it is done purging the old Filament and I confirm, that the Color is OK, then it goes back to the black screen.

Thank you for the continus support of my Prusa (formally an MK3, now MK4)

bkerler, 2023-12-21:
duplicate of #3525

closed 2024-06-05 by danopernis