prusa3d/Prusa-Firmware-Buddy #3589: [FEATURE REQUEST] PCTG as new Material

Keridos, 2023-12-13:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MK4

**Describe the expected functionality**
Add a new material profile for PCTG:
Nozzle temperature: 260°C
Heatbed termperature: 100°C

Two examples for this type of filament

bkerler, 2023-12-14:
PCTG seems to have the same Temp specs as ASA. Shall it still be added ?

bkerler, 2023-12-14:
I've added PR #3591 for this request. It would be good if you can confirm that you successfully printed with that material on the MK4.

Keridos, 2023-12-14:
I printed successfully. Have not fully worked out the retraction settings yet as my 3DJake sample was a bit to moist after shipping (suspect the packaging was faulty). But after drying it prints ok. A bit of stringing, but that's it.

dstulken, 2023-12-14:
> PCTG seems to have the same Temp specs as ASA. Shall it still be added ?

Yes.  It's much more straightforward from the user perspective.  And combining them ("ASA/PCTG" in the menus, etc) would be a strange as they are completely different filament compositions.   Leaving them as separate items, even if they have similar values under the hood, also allows the temps and profile characteristics to diverge over time if tuning and feedback suggest they print differently and do need different treatment.

bkerler, 2023-12-14:
OK, then the PR should be fine for that. PCTG is a separate item.

bkerler, 2024-01-02:
I added a new PR which allows usage of custom filament profiles, which is even more convenient :)

silvvero, 2024-03-30:
I would also like to requst PCTG as new filament in the mk4.
PrusaSilcer support it in mk4 profiles so the machine itself should also support it.

kvtorp, 2024-05-07:
Also the XL could use the ability to load PCTG. PCTG is available in PrusaSlicer, but not in the material load menu, leading to lots of warnings when sending a print as the materials do not match

CZDanol, 2024-06-01:
Hey, I will try to push for adding support for custom material types. Unifying the related issues under the internal ticket BFW-5513

CZDanol, 2024-09-11:
Hello, we've released a 6.2.0-alpha1 firmware with user material types support: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha1

This means that while we decided to not put PCTG in our presets, you can set it up yourselves :)

closed 2024-09-11 by CZDanol