prusa3d/Prusa-Firmware-Buddy #3592: [BUG] "Mass (W) must be greater than 0" error

jerrygrey, 2023-12-14:
**Printer type** - MINI+

**Printer firmware version** - 5.1.0

**Original or Custom firmware** - Original

**Describe the bug**
The M572 command in the start g-code for MINI+ in PrusaSilcer causes "Mass (W) must be greater than 0" error.

**G-code**
`M572 W0.06 ; set smooth time`

bkerler, 2023-12-14:
The error "Mass (W) must be greater than 0" is actually related to the M74 gcode command and not the M572 command afaiks: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/15d5ae0b317a37b69b8e7a4720711b962c25b57d/lib/Marlin/Marlin/src/gcode/feature/input_shaper/M74.cpp#L54

bkerler, 2023-12-14:
After testing the firmware, the error "Mass (W) must be greater than 0" does only appear with M74 gcode command if the value is 0. The Smooth time (between 0 and 0.2) would instead indicate the error "Pressure advance smooth time (W) value out of range (0-0.2)". I don't see any relation between M74 and M572 in the code to be honest ...

jerrygrey, 2023-12-14:
@bkerler I double checked and you're correct the two are not related.

jerrygrey, 2023-12-14:
Updated, related [PR#3594](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3594)

bkerler, 2023-12-15:
I've updated the PR and can confirm that prusaslicer indeed ALWAYS adds "M74 W0" with IS, so the PR does make sense except if prusaslicer shouldn't set the weight to 0 at all.

bkerler, 2023-12-15:
It's now merged so this issue can be closed :)

closed 2023-12-15 by jerrygrey