prusa3d/Prusa-Firmware-Buddy #3593: USB filesystem is corrupted on model upload

lggomez, 2023-12-14:
I've been uploading several files during the day (no shutdown or reboot) until an upload returned an error and the pendrive was rendered useless until I reformatted

IIRC the error was along the lines of file not found (unfortunately I didn't take a screenshot or request log from the browser), but FWIW I can share the file that triggered the corruption: [6x6906_0.4n_0.15mm_PLA_MK4IS_3h36m.zip](https://github.com/prusa3d/Prusa-Link/files/13665773/6x6906_0.4n_0.15mm_PLA_MK4IS_3h36m.zip)

This happened on an MK4 with 5.1.0 firmware, input shape on

TojikCZ, 2023-12-14:
Hi, I have transferred your issue to the buddy firmware repo. The PrusaLink one just deals with MK3 and older support through raspberry pies

Zappes, 2023-12-14:
I have encountered a very similar problem and the simple explanation was that the USB stick that came with my MK4 was cheapocrappic and thus broke. I have replaced it with a Sandisk stick and since then, no further issues have come up.

eWuPi, 2023-12-15:
I also had such a problem, I used HP in my mini and it was ok, when I translated to mk4, it also started to jam and only the reset printers helped. I put on a different memory and for now it is OK.
Dodakowo the memory problem caused a failure of the Web Prusalink interface, the API was active.

lggomez, 2023-12-15:
I had a second instance of file corruption less than a day after the first one so I assume that the stick is pretty much DoA (it lasted less than 2 weeks):

[image]

Very disappointing, aside from the memory quality as @Zappes says the stick had a nice finish

But anyway, I'm using a new one and expect to not have further issues

bkerler, 2023-12-16:
That's most likely an issue with the usb drive or the way you stored data on it (remember to always eject the stick on the pc otherwise the fat filesystem will break very fast and only repair will help). Not really a firmware issue or bug.

m-cas, 2023-12-20:
@bkerler I disagree, I had twice the Prusa stick go blank (read: empty) while being used on the printer. And I am an IT professional the rather eject the sticks twice than to just yank it out of the usb cable... not the most impressive quality but easy to fix: use a better (and much faster too) stick, nowadays they cost nothing...

bkerler, 2023-12-20:
@m-cas so you disagree to agree :) Anyways glad you fixed it, but still it's no fw bug.

42degrees, 2024-01-09:
I've just had a very similar thing happen.  First, Saturday (two days ago), it said there were files on the USB, but gave errors when trying to delete one.  I put the USB in my computer and it had only the last bgcode file that was printed on it (where before it had all the sample files and bgcode files we had ever printed).  So, I reformatted the drive Fat32 (just to eliminate the possibility of corruption of the file system), then we did a print (100% success). For some reason the connect site still said all the files were on the drive, but since I formatted it, that's not possible.

Then after uploading a big bgcode file (33MB) the connect dashboard started saying that the USB drive had no files on it.  I put it in my computer and it tells me the drive must be formatted before use.  I put it back in the printer and it says there are no files on it.  The first problems occurred on previous firmware version (presumably 5.1.1?), the second happened after upgrading to 5.1.2+13478.  The drive is now showing that the primary partition is RAW.  I'm going to format it again and hopefully it will allow us to print.

As of this time my current print is going just fine after re-formatting yet again, copying the bgcode file on there and putting it back in the printer.

danyev3, 2024-06-09:
I ran in a similar issue. It's about the third time my USB (stock one that came with MK4) got corrupted and I lost all files on it. Please note I NEVER unplug my usb from my printer, I only upload files via prusa connect (printer is connected via a lan cable), this time the printer was printing, it finished the job, i tried to upload a new file from prusaslicer via connect, and the upload failed, upon reloading the page all files on the printer disappeared, I plugged the usb in my pc and it said it needed to be formatted to be usable. testdisk couldn't recover any files. pretty disappointing as I had some pretty complex to set up jobs saved on there that were yet to be printed

danopernis, 2024-09-02:
Hi folks, while I understand your frustration, due to the randomness of this bug it is very difficult to fix. Without clear reproducer, we can't really work on this :disappointed:

bkerler, 2024-09-02:
@danyev3 thanks, this helps to narrow down the issue. Question: Is your printer running all the time or do you power off the printer after the prints have finished ?

danyev3, 2024-09-02:
> @danyev3 thanks, this helps to narrow down the issue. Question: Is your printer running all the time or do you power off the printer after the prints have finished ?

Hi, thanks for taking the time to look into this issue. I cut power to my printer whenever i'm not printing or i'm not using my pc (they are both connected to the same switched outlet). Let me know if i can provide more details.

bkerler, 2024-09-02:
I had similar issues recently but that was due to the outlet strip causing some power spikes when turning the printer on.
I assume that either it is a similar case or the problem might be within the FAT implementation. FAT is very fragile when files are still buffered and the device is powered off before writing as FAT has no journaling system similar to NTFS or Ext4.

To rule the power issue out: Did the error happen before the first print or midst a print ?

tomekziel, 2024-10-27:
Same issue just happened on my Prusa Mini+, sudden pendrive death (the one that came with the printer). Pendrive was permamently attached to the printer, I only used a Prusa Link to print. Printer was always powered down between prints.
Firmware 6.0.0+14794, bootloader 2.3.4

JeremiahGillis, 2024-10-31:
I just had this issue occur when using Prusa Connect through PrusaSlicer on my XL. My USB drive was corrupted. I had to format the drive in my computer. I am using the included Prusa Flash drive and I never remove it. My model was size was < 3MB.

Firmware: 6.2.0-alpha2+8702
Prusa Slicer: PrusaSlicer-2.8.1+win64-202409181359

Halfino, 2024-12-12:
I had this exact issue today, formatting USB didnt resolve the issue, when I wanted to download connect files to the printer, it corrupted the USB data twice in a row after formatting. Uploading gcode from PC to USB works, but is annoying.

Firmware: 6.1.3+7898
Slicer: Prusa Slicer 2.8.1+win64

murk-sy, 2025-01-12:
Happened to me as well, it made multiple folders for the upload, corrupted the USB drive name and all print files on it as well. There was a MSDOS file there which I assume is a string from the filesystem.

Formatted the drive using quick format in windows with default settings. Reinserted into the printer. All the files were back but there was an error on insert. Okay then.
Plugged into windows. Files were back but there was an error. Repaired the error, which made some files disappear. Stuff that disappeared in windows reappeared on the printer.
Ran a full format for a few seconds to get rid of all the file pointers and then did a quick format again. Finally the USB drive is clean on the printer as well. Upload via connect appears to work again.

XL 5 tool
Firmware: 6.2.0-RC1+8787

gazz29, 2025-01-17:
**Try using Prusa Link to send the files?** 

I have a XL5t and a MK3.5, 
about a week after i had built / upgraded the printers, i replaced the USB sticks with physically smaller and lighter sandisk ones due to getting similar errors people mention, i was thinking the weight of the metal Prusa USB stick may be causing the issue... it wasn't, i still got loads of USB Read / Write errors.

_I cured my USB **Read** errors (mid print stoppages etc) by fitting a couple of ferrite cores around the screen to main board ribbon cable at each end.... 
 i haven't had a single **USB Read** error since i did this in May 2024.
My theory is EMI is interfering with the data going down that unshielded ribbon cable when the printer is in motion... i've made many posts about this on facebook groups, and posted a video demonstrating ferrite cores curing EMI by showing my milling machines DRO speed reading going crazy, then i clip a ferrite core to the speed sensor lead (it's an inductive pickup, a bit like a pinda probe) and the speed reading becomes rock solid and stable._  

Since fitting the ferrite cores to the screens ribbon cable, i haven't had a SINGLE **USB Read** error for over 8 months on either printer

.
I have however had lots of USB **Write** errors when sending the files from Prusa Slicer to the printers using Prusa Connect, 
I figured the data is getting corrupted on the way from my laptop out to the www, to the servers and then back to my printer, 

So 7 days ago i switched to using Prusa Link to send the files from Slicer to the printers as an experiment. 

I haven't had a single USB **Write** error when using Prusa Link to send the files over my home network, obviously using Link removes the storage of your files on the cloud, so printing from anywhere doesn't work etc, but i am using my printers for home hobby use, so this is fine. 
.

However, i found that i need to sometimes press the 'Set Ready' button in Connect to tell the printers the print sheet is 'empty and clean' so it will start a print i've sent using the 'Upload and Print' command in Slicer, when using Link to transfer the data. 

So today i switched back to using Connect to send the data over, as this pops up a message asking "Is the printer ready? Is the print sheet in place, empty and clean?" and clicking OK starts the print. 

But of the 10 files i sent the printer today using Connect, 4 of them failed with USB **Write** errors!
So i'm back to using Link again for now, and sent the same files via Link and they printed perfectly.

bkerler, 2025-01-25:
> 
> A detail that may help: There were **multiple** folders with either similar or the same name of the last upload (to be clear i tried to start the print 2-3 times before realizing something was wrong, but there were definitely more folders than upload attempts), and the prusa_uploads.txt file was intact.
> 
> XL 5 tool Firmware: 6.2.0-RC1+8787

That's interesting. I think I saw that the file functions are thread-safe, but I will look into that.

bkerler, 2025-01-25:
> Since fitting the ferrite cores to the screens ribbon cable, i haven't had a SINGLE **USB Read** error for over 8 months on either printer

That clearly sounds like you have a serious grounding / PSU issue. A ferrite core really isn't needed except you built and use custom cables than the stock ones.

Regarding the connect write issue: Are you sure it is a usb write error and not that the file hasn't been sent at all ? Connect won't send the file if the same filename already exists ..., that's why I use {timestamp} in my printer output settings in the filename.

gazz29, 2025-01-26:
>That clearly sounds like you have a serious grounding / PSU issue. A ferrite core really isn't needed except you built and use custom cables than the stock ones.

who's using custom cables? 
i'm using the stock unshielded ribbon cables that came with the printers to link the screen and usb port to the main board. and the stock psu's that came with the printers, stock grounding cables between the screen and printers etc,  it's the same on my MK3.5 and my XL, 

What i do know is that i have not has one single USB Read error on either printer after i fitted the ferrite cores around the ribbon cable, the one that carries the usb data to the mainboard, along with the data for the screen and touch interface. 
I've not had one single mid print stoppage either, or any of the other issues people post about all the time relating to this USB Read error. 

4 x clip on ferrite cores turned my printers from being semi unreliable to being rock solid, for a cost of less than £10 (they came in a pack of 5, so i put a core at each end of the ribbon cable, which is quite long on the XL)

.
**I really think prusa should be using a shielded cable between the screen and main board,** but even shielded cables can pick up EMI (i can demonstrate this with my milling machines DRO, the speed reading goes crazy showing from zero to 180k rpm until i clip a ferrite core to the _(shielded and grounded at the star grounding point of the DRO head)_ inductive pickup / speed sensor wire, then instantly it's rock solid and reading the true 90 - 94 rpm the spindle is turning at)

:

> Regarding the connect write issue: Are you sure it is a usb write error and not that the file hasn't been sent at all ? Connect won't send the file if the same filename already exists ..., that's why I use {timestamp} in my printer output settings in the filename.

Many times i was able to go to my printer and press ok to the 'USB Write error' message on the screen, then select 'print' and there is the file it claimed wasn't sent, and it prints out perfectly!

Then with an update to Prusa Connect's browser, i could press 'OK' on the red message that drops down when this was happening, and sure enough the file is on the printers USB stick, and i could press print and away it goes and completes the print perfectly. 

However i have gotten fed up of this happening, So i am staying with using PrusaLink to send the files to my printers instead of Connect (i still monitor my printers using Connect)
Since using Prusa Link to send the files i have not had one single USB WRITE error, and i've sent over 500 prints to my printers this past week alone (some of them 9 hour prints, some 4 minute prints where i am doing a manual colour change to make tiny switch label plates on my MK3.5 with a 0.15mm revo nozzle installed.

closed 2025-04-15 by github-actions[bot]

JeremiahGillis, 2025-04-15:
This should not be closed until this is fixed. Filesystem corruption occurs with PrusaLink (this issue) and Prusa Connect (#3706).

Cdddo, 2025-11-18:
Just ran into this issue. Was uploading a file to print and it failed. I tried to reupload the file and it failed again. So I looked into the files on the printer and there were a bunch of corrupted folders and files. I had been printing on the machine just fine minutes before.

[image]

Note that I never take the USB off the machine. This seems to be an issue with prusaconnect or the firmware.

MK4
Firmware: 6.4.0-RC+11796
PrusaSlicer: 2.9.4