prusa3d/Prusa-Firmware-Buddy #3596: [BUG] <Enter comprehensive title>

danopat, 2023-12-15:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI+

**Printer firmware version** - 5.1.0 and 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
 USB

**Describe the bug**
I have 2 prusa mini printers, both refused to connect to the wifi network after update from fw 4.xx to 5.1.0, before the  update it worked OK. Ethernet works OK. Different wifi network also doesn't work. Tried uploading new wifi drivers and network credentials both generated by printers and by slicer like a hundred times. The wifi module led light shines red. 

After some meaningless clicking on some network settings one of the printers successfully connected and is now able to connect within seconds after reboot/firmware update to 5.1.2. Is there any hidden functionality or is it just a coincidence? No success with the second printer. 

Update to 5.1.2 didn't help. The chatsupport suggested to open an issue here.
[image]
[image]
[image]

**How to reproduce**
Update prusa mini from 4.xx to 5.1.0

**Expected behavior**
Functional wifi connection

bkerler, 2023-12-16:
Happened to me as well. You can write a new config to the usb stick and then set up the wifi details and then load it in the settings menu. Worked for me. If the esp bootloader wasn't flashed correctly, best downgrade to 4.7.5 and then upgrade to 5.0 first before upgrading to 5.1.2.

danopat, 2023-12-16:
> Happened to me as well. You can write a new config (store credentials) to the usb stick and then set up the wifi details and then load the credentials in the settings menu. Worked for me. If the esp bootloader wasn't flashed correctly, best downgrade to 4.7.5 and then upgrade to 5.0 first before upgrading to 5.1.2.

Will try the downgrade, thanks!

danopat, 2023-12-16:
UPDATE: Downgraded to 4.4.1 and the wifi worked, after upgrading to 5.1.2 it still works so it must have been something bad in the fw upgrade.

bkerler, 2023-12-17:
Great to hear. So I assume this issue can be closed ?

szamelsz, 2024-03-24:
Happened to me also, for me it worked after first upgrade to 5.1.2, but I needed to downgrade back to 4.4.1 - at this point the wifi works after configuration. Then updated back to 5.1.0 -> 5.1.2 - now it doesnt connect to the same network as before. Also tried to configure it many times.

closed 2024-07-01 by github-actions[bot]