prusa3d/Prusa-Firmware-Buddy #3597: [BUG] Error! #13604 when powered on

dp250f, 2023-12-16:
**Printer type** - MK4

**Printer firmware version** - 5.1.2 release

**Original or Custom firmware** - Custom (#define HEATER_0_MAXTEMP 320)

**Optional upgrades** - Micro Swiss CM2 RepRap V5-V6, silicone hot end sock

**USB drive or USB/Octoprint**
  PrusaConnect

**Describe the bug**
  Error! #13604 displays when powered on.
  Printer boots normally when reset with reset button.

**How to reproduce**
  Flash 5.1.2 firmware. Turn off printer. Turn it back on.

**Expected behavior**
  Printer should boot normally

**G-code**
  N/A

**Crash dump file**
  "No crash dump to save"

**Video**
  Please attach a video. It usually helps to solve the problem.

bkerler, 2023-12-16:
Haven't tested the 5.1.2 yet, but Error #13604 means : "Firmware file missing in the USB flash!". It could be that the file wasn't written completely or the filesystem could be broken. Also make sure you don't have several fw upgrades in the same root directory on the usb stick.

bkerler, 2023-12-17:
Version 5.1.2 says that this issue should be fixed ... so maybe this version introduced an issue instead.

dp250f, 2023-12-18:
Reformatted the USB drive. Still displays error on cold boot and not on reset.
I also tried without the USB drive and it boots normally.

Other than the error message on cold boot, everything works normally.

dp250f, 2023-12-30:
Bought a new flash drive, formatted FAT32. Still the same thing happening.

dp250f, 2024-05-03:
Built and flashed 6.0.0 release today. Issue persists.

closed 2024-07-15 by github-actions[bot]