prusa3d/Prusa-Firmware-Buddy #3600: [ENHANCEMENT] Add "nozzle diameter" as an optional value to display in the User Interface Footer

RCNet46, 2023-12-17:
**Printer type** - MK3.9
**Firmware** - 5.1.2

**Is your suggested improvement related to an existing problem? Please describe.**
This is an enhancement to add the "nozzle diameter" to the list of options of items that can be selected and displayed in the User Interface Footer. With the ability to easily change the nozzles with the Nextruder or with an E3d Revo on the mini, this enhancement would allow the value of the current configured nozzle diameter to be displayed in the User Interface Footer, and would provide a quick "sanity check" that the printer is configured as expected before starting a print, or requiring the user to navigate the menus to confirm the nozzle diameter setting.

**Describe the expected functionality**
Currently there are a lot of useful values that can be displayed in one of the five fields of the User Interface Footer as configured under Settings->User Interface->Footer->Item 1-5.  Add an additional option [Diameter] which displays the currently configured nozzle diameter value that was set in Settings->Hardware->Nozzle Diameter.

bkerler, 2023-12-17:
I've just added your feature request as a PR. Enjoy.

danopernis, 2024-06-06:
This is fixed since v5.2.0 via commit https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/954734967c8811c4241e4dc9a178961ccc388be6

closed 2024-06-06 by danopernis