prusa3d/Prusa-Firmware-Buddy #3603: [BUG] cool down button brakes temperature number

PoliEcho, 2023-12-17:
**Printer type** - [MINI+]

**Printer firmware version** - [5.1.2+13478, 5.1.0; language: cz]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [none]

**Describe the bug**
  If you press cool down button and temperature set has two and more digits zero will be just put over the number.
  (see attached video)

**How to reproduce**
  Set temperature to temperature with two or more digits than press cool down.

**Expected behavior**
  Zero will fully replace the number.

**Video**
[fw_issue.webm](https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/138225515/7bffb03a-0a90-4556-b388-296f45a7df1b)

bkerler, 2023-12-18:
Looks like a display refresh issue/bug. I will have a look if this also happens on the MK4.

bkerler, 2023-12-18:
Ok, I can confirm that this bug only happens with the Mini. It can be reproduced as well in the simulator.

[image]

Seems like the lcd line isn't fully refreshed/cleared.

closed 2024-08-13 by github-actions[bot]