prusa3d/Prusa-Firmware-Buddy #3605: [BUG] Cannot set nozzle temperature higher than 275°C

d4nix-cz, 2023-12-18:
**Printer type** - [MINI]

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB flash drive or USB

**Describe the bug**
  When I had FW version 4.4.1 the Mini let me set nozzle temperature to at least 290°C. From version 5.1.0 I'm able to set temperature only at least to 275°C, no matter if I try it from menu or use temperature set in gcode.
As Prusament PC Blend Carbon Fibre filament is recommend to print at 280°C then it's now not possible to print this material on Mini. I've printed several spools of PCCF at 288°C on my Mini in the past without any issue.

**How to reproduce**
  Go to menu > Control > Nozzle > set temperature as high as possible

**Expected behavior**
  Mini should let me set nozzle temperature to 290°C

**G-code**
  _not relevant_

**Crash dump file**
  _not relevant_

**Video**
[link to video](https://photos.app.goo.gl/rymZmoFHrioNbLFc7)

bkerler, 2023-12-18:
Actually the mini heater max temperature is 290°C and the decrease is due to a satety margin of 15°C. Not really a bug but a safety requirement in order to prevent components from overheating.

Th3Rom3, 2023-12-18:
Still there is aa small conflict since the MINI is advertised with 280°C max nozzle temp:

https://www.prusa3d.com/product/original-prusa-mini-semi-assembled-3d-printer-4/

Some official clarification/comment  would be nice.

kozross, 2023-12-18:
[Duplicate of 3489](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3489). 

@bkerler - this is wrong, and wildly so. This has never been a problem prior to the 5.0.0 line, and, as the linked issue specifies, this behaviour is in direct contradiction with _at least_:

* The Mini's product page
* The Mini's official documentation in at least one place
* The _rest of the Mini's firmware_ in at least one place (nylon loading routine)

It's a bug, not a feature.

bkerler, 2023-12-18:
@kozross That's just what the source code says, not my opinion. 

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/cf299fc7e84532f60ba8615074ea3d201701536c/include/marlin/Configuration_MINI.h#L406

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/cf299fc7e84532f60ba8615074ea3d201701536c/lib/Marlin/Marlin/src/lcd/extensible_ui/ui_api.cpp#L893

Patching that would be easy, but as it's a safety value, I'm waiting for a PR until further confirmation by the prusa developers if the safety margin can be lowered to a value of 5 or 10.

bkerler, 2023-12-18:
I've prepared a PR, but still waiting for a confirmation by Prusa.

bkerler, 2023-12-22:
PR #3635 has been merged and issue can be closed :)

d4nix-cz, 2023-12-22:
@bkerler ,
Sadly from my point of view the proposed solution (rise the temperature to 295°C to have max temp 280°C) is not acceptable, because my Mini+ printed at least 10 KG of material at temperatures between 285 and 290°C. I bought it as capable to print PC Blend and Nylons, now it's machine only usable only for general materials such PLA and PETG. 

So my machine lost its value!

And again, using lower version of FW is not a solution as new FW provide much better quality prints!

kozross, 2023-12-22:
> @bkerler , Sadly from my point of view the proposed solution (rise the temperature to 295°C to have max temp 280°C) is not acceptable, because my Mini+ printed at least 10 KG of material at [...]

Out of curiosity, what temperature(s) do the filament profiles for nylons (and PC Blend) for the Mini specify? If these are no longer achievable, then the Mini is still non-conformant.

d4nix-cz, 2023-12-22:
@kozross,
for example my favorite PC Blend Carbon Fibre have 280°C as default temperature for prusa mini.
[image]
And due to increased speeds in Input Shaper profiles the nozzle temperature have to be higher to let properly melt material for printing.
And here you can see recommended specs for that material - [https://www.prusa3d.com/product/prusament-pc-blend-carbon-fiber-black-800g-2/#printing-setup](url)

kozross, 2023-12-22:
@d4nix-cz - interesting! Definitely agree with you.

Also, fun fact: the Mini's [product page](https://www.prusa3d.com/product/original-prusa-mini-semi-assembled-3d-printer-4/) specifies that it is able to print nylons, but Prusament PA11-CF has no profile for the Mini with input shaping, probably because it requires 285C according to specs.

bkerler, 2023-12-22:
> @bkerler , Sadly from my point of view the proposed solution (rise the temperature to 295°C to have max temp 280°C) is not acceptable, because my Mini+ printed at least 10 KG of material at [...]

A temperature of 280°C is possible with the new commit. In fact it will fluctuate between 290°C and 280°C. Printing PA thus should be no problem (I was told on tests it worked fine). You can't expect more than 280°C as it was never being sold being able to provide 290°C. So you can still break the appendix and remove the safety limit and then flash your custom firmware. But if your printer will overheat or melt, it's your problem.

bkerler, 2023-12-22:
> @d4nix-cz - interesting! Definitely agree with you.
>
> Also, fun fact: the Mini's [product page](https://www.prusa3d.com/product/original-prusa-mini-semi-assembled-3d-printer-4/) specifies that it is [...]

Exactly. No fun fact, the page states 280 °C nozzle and 100 °C heatbed temperature for the PA11-CF. And that is achievable and PA is printable with that.

bkerler, 2023-12-22:
> @kozross, for example my favorite PC Blend Carbon Fibre have 280°C as default temperature for prusa mini. [...]

I doubt that for increased speeds you need a higher temperature. I used to print with and without IS at the same temperature with the same results. If you were saying flow rate ok. But higher temperature doesn't make any sense to me.

d4nix-cz, 2023-12-25:
> > @bkerler , Sadly from my point of view the proposed solution (rise the temperature to 295°C to have max temp 280°C) is not acceptable, because my Mini+ printed at least 10 KG of material at temperatures between 285 and 290°C. I bought it as capable to print PC Blend and Nylons, now it's machine only usable only for general materials such PLA and PETG.
> > So my machine lost its value!
> > And again, using lower version of FW is not a solution as new FW provide much better quality prints!
> 
> A temperature of 280°C is possible with the new commit. In fact it will fluctuate between 295°C and 280°C (the so called safety margin of 15, which will max out at 290°C). Printing PA thus should be no problem (I was told on tests it worked fine). You can't expect more than 280°C as it was never being sold being able to provide 290°C. So you can still break the appendix and remove the safety limit and then flash your custom firmware. But if your printer will overheat or melt, it's your problem.
> 
> And no, your machine hasn't lost its value. You can still use the old firmware. Input shaper was a bonus that was never advertised when you bought the Mini.
> 
> Here's a pic using the simulator for the commit, testing PA: ![image](https://private-user-images.githubusercontent.com/6317772/292590314-9b71e330-4e55-4ed4-b82d-52544b94ccb6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM1MDMxNjksIm5iZiI6MTcwMzUwMjg2OSwicGF0aCI6Ii82MzE3NzcyLzI5MjU5MDMxNC05YjcxZTMzMC00ZTU1LTRlZDQtYjgyZC01MjU0NGI5NGNjYjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjVUMTExNDI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzExYjFhYmE3NjMwNzgxODEzNjI3YzRhM2UzY2Q0MjEyYzA3ZDVkNDc2NmRlMGVmMWM0NzUyODFlNzJiOGE5ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.UVxBQ6upxpV5jQwIzKGrXIe1jfvXYH3Hp6TGNtnt0kM)

Please, can someone point me to page where I can learn how to create my own Firmware based on latest Prusa's FW?

Prusa-Support, 2024-01-17:
Information about the firmware building can be found in the repository's [read me](https://github.com/prusa3d/Prusa-Firmware-Buddy) (description) but firmware customizations shouldn't be necessary.

As mentioned earlier in the thread, the fix is already merged in the source code (PR https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3635), hence will be integrated with the next firmware release.

I'm sorry for the confusion this may have caused.

Michele Moramarco
Prusa Research

kozross, 2024-01-18:
@Prusa-Support - the loading temperature for nylon on the Mini is still 285. Will this get fixed as well?

Prusa-Support, 2024-02-17:
It was already discussed extensively so theoretically yes. However, AFAIK the printer will work regardless of having a temperature set above the maximum value but the temperature will stop at 280°C - I can't tell for sure as the next firmware is not available yet.
Sadly, there is no estimated time for the next firmware release at the moment though as they are polishing the firmware compatibility with MK3.5 - for more information, please check the [releases](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues).

Michele Moramarco
Prusa Research

kozross, 2024-04-12:
@qbranchco - I basically ended up getting rid of my Prusa Mini because of this issue. I can no longer trust Prusa, since it is now close to _five_ months since their latest firmware has made their own product incompatible with its own product page. I do not understand how cutting a release is so difficult, especially since the fix has already been mainlined, _especially_ over such a serious issue.

Supremely disappointed, and would _definitely_ aim people at this, and related issues, if ever asked if Prusa products are worth buying. Hint: they are not.

Prusa-Support, 2024-04-16:
After additional tests and reviews, the maximum allowed temperature on Prusa MINI is capped at 280°C for safety reasons, and in accordance with the product description.

The incorrectly set maximum temperature of 275°C has been fixed in [FW 6.0.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0). 
The user can heat up the nozzle up to 280°C without any worries, and this should be the final resolution that closes this issue.

Michele Moramarco
Prusa Research

closed 2024-04-16 by Prusa-Support

qbranchco, 2024-04-16:
Thank you Michele Moramarco for intentionally crippling thousands of users devices they've purchased from Prusa, and making my Revo nozzle (E3D also being your technology partner for Nozzles, no less!) useless.

What tests?
What reviews?
What safety issues?
Why is this not being disclosed publicly?

If this project is really open source, in the spirit of open source the data, solid reasoning and explanation should be made here, publicly.

lumen-novum, 2024-04-16:
@qbranchco I have the Revo Micro and can compile a build for you and anyone else who needs the max temp of the revo. ;)

I definitely agree with you though, Prusa should be more transparent with a device that has the potential to start a serious fire. This is not the first time Prusa has failed to deliver on their promises (quality control, extruder issues, power panic, etc.). I hope they pull themselves together and stop letting customers down when their product clearly needed more time in oven.

kozross, 2024-04-17:
> @qbranchco I have the Revo Micro and can compile a build for you and anyone else who needs the max temp of the revo if you'd like. ;)
> 
> I definitely agree with you though, Prusa should be more transparent with a device that has the potential to start a serious fire. This is not the first time Prusa has failed to deliver on their promises (quality control, extruder issues, power panic, etc.). I hope they pull themselves together and stop letting customers down when their product clearly needed more time in oven.

It's far from the first time where the Mini is concerned. For instance, I know _for a fact_ that the Mini over-draws its PSU on the regular (because I measured), and Prusa's claims regarding the power usage of its bed are _straight-up_ untrue as a result. This hasn't even been acknowledged, much less addressed anywhere. I'm not even going to mention an issue causing [underextrusion and jamming](https://forum.prusa3d.com/forum/general-discussion-announcements-and-releases/avoiding-early-prusa-mini-issues-common-issues/) (which I and countless others have experienced) which has been around _since day freaking 1_: also completely un-acknowledged and un-addressed, despite being replicatable. Or its _utter_ lack of cooling due to a weird 5V fan fixation and awful fan duct design?

The Mini is a product Prusa do not care about, and never have, and it's sad.

d4nix-cz, 2024-04-18:
> @qbranchco I have the Revo Micro and can compile a build for you and anyone else who needs the max temp of the revo if you'd like. ;)
>
> I definitely agree with you though, Prusa should be more [...]

If you know how and able to compile it, please do so. My mini print fine now with 280°C, but as I have Revo too, I would like to have higher nozzle temperature for sure - as my Mini prints fine at 290°C for two years and without any issue.

qbranchco, 2024-04-19:
I don't, and to do so I would need to break the tamper-proof seal on my Prusa Mini's buddy board, voiding any warranty, and running unsigned, unverified firmware.

It would be a better solution for ALL Prusa Mini owners for Prusa to be transparent about the tests they performed, the safety issues they claim, for the information to be verified by the community and a decision made in the open. Perhaps an option to select for Revo nozzles where higher temps are allowed with a higher wattage PSU?

lumen-novum, 2024-04-22:
For those of you who wish to install the custom firmware [here it is.](https://github.com/lumen-novum/Prusa-Firmware-Buddy/releases/tag/v6.0.0-hotter-hotend)

I unfortunately can't test the reliability of the firmware while printing at 295°C since I don't have high temperature filaments nor do I have a proper setup to do so. However, I can confirm my hotend was able to get up to 295°C for a couple of minutes.

If anything, this is certainly a band-aid solution and I hope Prusa addresses this issue. In the mean time, I hope this helps someone out. 🙂

d4nix-cz, 2024-04-22:
> For those of you who wish to install the custom firmware [here it is.](https://github.com/lumen-novum/Prusa-Firmware-Buddy/releases/tag/v6.0.0-hotter-hotend)
> 
> I unfortunately can't test the reliability of the firmware while printing at 295°C since I don't have high temperature filaments nor do I have a proper setup to do so. However, I can confirm my hotend was able to get up to 295°C for a couple of minutes.
> 
> If anything, this is certainly a band-aid solution and I hope Prusa addresses this issue. In the mean time, I hope this helps someone out. 🙂
> 
> Please let me know if this firmware works for you guys or if I need to adjust some things.

Thank you so much for your effort. I hope that someday this week I find courage to install it and test it! =)