prusa3d/Prusa-Firmware-Buddy #3606: [ENHANCEMENT] Show remaining time all the time

espr14, 2023-12-18:
**Printer type** - [MK4]
**Optional upgrades** - [N/A]

**Is your suggested improvement related to an existing problem? Please describe.**
I have to wait to see time remaining because it shares the same space with other times.

**Describe the expected functionality**

[image]

Tupson444, 2024-06-07:
There is plenty of space on the screen so that all the useful information can be shown at once. I want to be able see the remaining time at a glance, without any waiting.

Currently, the larger screen has almost no advantages over the old MK3 screen, because it is not used properly. The same amount of information in same font size could be shown on a much smaller screen.

Tupson444, 2026-09-28:
still waiting for it