prusa3d/Prusa-Firmware-Buddy #3609: [BUG] Error! #12604 (firmware file missing) when MINI+ powered with some USB flash disk

TwadaCZ, 2023-12-18:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
MINI+

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.2 (latest)

**Original or Custom firmware** - [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB flash drive

**Describe the bug**
When restart printer or power on printer... and USB disk is present... after bootloader 2.0.4 printer show red screen with Error 12604 - Firmware file missing on this USB drive - with "continue" button, but it is not working. At previous firmware 5.0.1 it was possible to click continue and printer started.

Now at 5.1.2 is not possible to "click" to continue.

Printer starting normally, if USB drive not present during start. Or when I put firmware .bff file to USB drive, it starting properly too.

With some USB drives this error not shown... I do not know why.

**How to reproduce**
Restart (PowerON) printer with USB drive inserted in.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**ScreenShots**
[image]
[image]

EllaBarker, 2023-12-19:
It's a known bug with 5.1.0: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.2
Also can confirm that this bug is still in 5.1.2.

> Error 13604 fixed
> 
> A bug found in the firmware release 5.1.0 affects MINI, MK4/MK3.9, and XL printers. If a USB drive is connected during startup, the printer may keep asking for a firmware file (.bbf) and display error message #13604. This issue has been identified and fixed in the current release.

TwadaCZ, 2023-12-19:
Yes, I know it... bud on my MINI+ still have this issue after upgrading to 5.1.2

EllaBarker, 2023-12-19:
I have the same issue with 5.1.2 (to be precise the bootloader). 

Haven't looked into the changes they made but the fix might be insufficient.

Gisi0, 2023-12-19:
duplicate of #3333

Koder22, 2023-12-24:
I found a solution to this problem on my Mini, i down graded to 4.4.1 and then changed the setting "FW UPDATE" to "off". Then i upgraded again to 5.1.0 and 5.1.2, and can boot without the firmware files.

[image]

arbok, 2023-12-25:
Reset to default and calibration solved the issue

muhendees, 2024-01-02:
Digging around a bit and found that this part checks if `sys_fw_update_is_enabled()` -- > https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/4bea923810302d654b932291ba324eaedff072fc/src/gui/screen_menu_fw_update.cpp#L25

This by default sets it to ON. I didn't found any other place where it resets after `v5.1.0` or anything related because new versions does not provide configuration options to enable or disable `FwAutoUpdate`  from GUI. Instead new versions will show you screen with QR code saying "To learn more including firmware downgrade, please visit: prusa.io/mini-firmware" for example here: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/4bea923810302d654b932291ba324eaedff072fc/src/gui/screen_help_fw_update.cpp#L78C2-L78C2

I have 2 assumptions:
- Removing this configuration change is intentional, (which is more reasonable to me) and probably transitional so it will be removed in the future(after following some traces on this front of changes on git logs) -- which means there was some mistake or bug during testing as I believe testing was done on board that didn't had this option enabled previously like in v4.4.1
- Also, if this option is not needed, update from `v4.4.1` to `v5.x.x` shoul reset this option and set it to `Off` but we see that is On by default

I would like to reset that option by feeding configuration file from USB - as I think it might be easiest option for me. If not possible, maybe recompiling with Option disabled by default and re-flashing FW so I can experiment with it. Going to read Contribution docs and see if I am comfortable in this rabbit hole :D 
Trying to see other options rather than reset/recalibrate

If I am totally wrong please correct me and I apologize for your time.

Prusa-Support, 2024-05-03:
Thank you for bringing this issue to our attention but please allow me to close this issue as duplicate.
Feel free to follow issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3333 or take part in the existing conversation.

Michele Moramarco
Prusa Research

closed 2024-05-03 by Prusa-Support