prusa3d/Prusa-Firmware-Buddy #3610: [ENHANCEMENT] Mini+ Firware Update: Amend prerequisites for 5.1.2. in documentation.

Th3Rom3, 2023-12-18:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** Prusa MINI+

I suggest amending a section in the documentation of new firmware releases (major and minor) to consistently and directly name the minimum bootloader and/or firmware requirements for an update.

In this specific example I was trying to pull up the firmware of a MINI+ from an old 4.3.2. version to the current 5.1.2. release.
While the release page for 5.1.0. mentions the requirements for a minimum bootloader version, the section is absent in 5.1.2. documentation.

I believe it would be worthwhile to always have those requirements listed, even with minor release versions to keep it consistent.

If I have missed something obvious, please disregard this issue.

Regards,
Tom

dstulken, 2023-12-20:
I'll second this.  A very worthwhile suggestion.

Th3Rom3, 2023-12-28:
Also ties into #3555

closed 2025-10-22 by github-actions[bot]