prusa3d/Prusa-Firmware-Buddy #3615: [BUG] File uploads from prusaconnect wipe & corrupt usb drive.

Ealiank, 2023-12-18:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Prusa MK4

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
PrusaConnect

**Describe the bug**
When uploading several files to the disk, the disk fails, corrupting it and stopping the print without a way to resume.

**How to reproduce**
During a print, slice & send over prusaconnect 5-10 files.

**Expected behavior**
An error saying the disk is not usable/readable.

**G-code**
Unobtainable.

**Crash dump file**
Unobtainable.

**Video**
Unobtainable.

hazelybell, 2024-01-06:
This happened to me when uploading a large file from PrusaSlicer. Also MK4 on 5.1.2.

samsmithnz, 2024-02-06:
This happens to me if the Prusa Connect files are at capacity. It throws a message to the user that you've reached capacity, and then corrupts the USB stick. I've replicated this 3 times.

katox, 2024-04-16:
The card can also be sometimes corrupted by removing other (unrelated) print files from the SD card during printing using PrusaLink.

samsmithnz, 2024-04-16:
This is a prusa connect issue, not prusa link.

alphen87, 2024-05-15:
Also just had this issue while uploading multiple files at once. I don't think the USB stick can be recovered. Also had this issue before when i was deleting files via connect.

closed 2024-07-23 by github-actions[bot]

samsmithnz, 2024-07-23:
Pretty disappointing this was auto-closed with no Prusa acknowledgement of an easy to replicate bug.

JeremiahGillis, 2024-10-31:
There is still a USB drive corruption bug posted here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3593