prusa3d/Prusa-Firmware-Buddy #3616: [BUG] Firmwear missing bug persists

JackpwMurray, 2023-12-19:
**Printer type** - [MINI]

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - [Original]

**Optional upgrades** - None

**USB drive or USB/Octoprint** - USB drive

**Describe the bug** - when booting the printer it shows an error: Firmwear file missing on drive, however unlike previuse versions, you cannot procide to startup

**How to reproduce** - Boot printer with an empty USB drive inserted

**Expected behavior** - Printer freezes on error screen.

bkerler, 2023-12-19:
This is a duplicate of issue #12604

Gisi0, 2023-12-19:
also of #3333

Prusa-Support, 2024-05-03:
Thank you for bringing this issue to our attention but please allow me to close this issue as duplicate.
Feel free to follow issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3333 or take part in the existing conversation.

Michele Moramarco
Prusa Research

closed 2024-05-03 by Prusa-Support