prusa3d/Prusa-Firmware-Buddy #3619: [BUG] MK4 does not start print via device, hangs on black screen, works via Upload &Print

MyChaOS87, 2023-12-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.1.12

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint**
Flash Drive

**Describe the bug**
After Starting Print, screen goes black except for header, only reset possible... Same Gcode Uploaded and started via Prusaslicer 2.7.1 works
 
**How to reproduce**
Start attached bgcode or gcode equivalent via on printer menu

**Expected behavior**
Printer starts with heat up routine and starts printing afterwards

**gcode**
[Hampelmann_Front_0.4n_0.2mm_PLA,PLA,PLA,PLA_MK4IS_2h38m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13718657/Hampelmann_Front_0.4n_0.2mm_PLA.PLA.PLA.PLA_MK4IS_2h38m.zip)
bgcode in zip file

bkerler, 2023-12-21:
Related to issue #https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3525

3D-Medic-Vince, 2024-02-29:
I've also found that you can start the print with the start print command via Prusa Connect and it will work when it won't start via the printer. I just started doing multicolor via virtual extruders and had this issue. I haven't tried starting via Prusa Link.

magoldsm, 2024-03-06:
Tried this.  It only printed in a single color.  Never paused for change.

bkerler, 2024-03-06:
As stated, it's issue #3525 and already fixed internally afaik. Most probably part of MK4 5.2 fw release.

ErickWarden, 2024-05-04:
> I've also found that you can start the print with the start print command via Prusa Connect and it will work when it won't start via the printer. I just started doing multicolor via virtual extruders and had this issue. I haven't tried starting via Prusa Link.

Exact same thing happened with me. It was only bgcode that hade two virtual nozzles. I thought it had something to do with the custome tool change code I was adding, BUT if I started the print via the prusa link it worked fine.

closed 2024-06-05 by danopernis