prusa3d/Prusa-Firmware-Buddy #3620: Random Filament changes during print now on 5.1.2, also present during 5.1.0[BUG] <Enter comprehensive title>

b4de4nd, 2023-12-19:
So ive run into a couple of random filament changes during some prints. Its not been a lot, but now ive had 2 changes in the same print randomly at both 16% and at 24% into the print.

Sliced with Prusaslicer 2.7.1 to printer MK4 on firmware 5.1.2 sliced with generic petg profile, MK4 IS 0.4 profile standard and structural 0.2 layer. Sliced into the new Bgcode and send via prusalink to the printer and print started on the printer it self.

This happened a couple of times on the 5.1.0 firmware also, but never more than 1 time on a print.

mix579, 2023-12-20:
Sounds a lot more like an issue with the filament sensor than a slicer bug?

bkerler, 2023-12-22:
Doesn't sound like a fw bug, as mix579 said, more like a filament sensor issue, maybe due to filament/plastic parts blocking the sensor.

danopernis, 2024-06-06:
Not a firmware bug

closed 2024-06-06 by danopernis