prusa3d/Prusa-Firmware-Buddy #3621: Prusaconnect in depth information - Need more info than printer needs your attention!

b4de4nd, 2023-12-19:
Hi

So I have a request that imo really needs to be implemented into prusaconnect. When you have a filament change, or a runout, or any other issue, the print is stopped and shows its on hold in prusaconnect. However the only message you can see is the printer needs your attention.

It would be awesome to implement that you would get information about the current state, also maybe in the overview, instead of showing paused it could show e.g "Filament runout" or "Filament change awaiting input" etc.

quasistoic, 2024-06-06:
Keeping alive...

dalbothek, 2024-06-06:
This has been implemented with [firmware 6.0.0](https://blog.prusa3d.com/major-firmware-release-6-0-0-prusaslicer-2-7-4-and-prusa-connect-updates-summary_95627/):

> The Prusa Connect team has added support for “Attention states” – essentially a much tighter communication between the printer and Connect. When a printer shows a status message on the screen, you have the same message displayed in Connect including active buttons which can be controlled remotely. Messages may also contain links to help pages related to the Attention state.

[image]