prusa3d/Prusa-Firmware-Buddy #3622: [BUG] Bad first layer calibration after one print after running calibrations & tests

sascape, 2023-12-19:
**Printer type** - Pre-assembled MK4

**Printer firmware version** - started in 5.1.0 and still exists in 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - nope

**USB drive or USB/Octoprint**
Both

**Describe the bug**
After running calibrations & tests successfully, I have a perfect first layer exactly once. If I try to repeat the print or try to print something else, the first layer is bad or doesn't even stick properly (see picture attached). If I run calibrations and tests again, the first print is perfect once again. Rinse and repeat.

**How to reproduce**
see above

[image]

[XMAS_Rentier_Lindor_Mod1_0.6n_0.32mm_PLA_MK4IS_1h23m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13720352/XMAS_Rentier_Lindor_Mod1_0.6n_0.32mm_PLA_MK4IS_1h23m.zip)

sascape, 2023-12-21:
I tried to change one thing at a time and I noticed that right now the first layer only looks like that with the textured print sheet. If I reprint the same file with the smooth print sheet, the first layer is perfect again...
Now to the fun discovery: If I then switch back to the textured sheet, the first print is fine again. Then it goes back to being as seen in the picture

So this is how it went:
1st print (textured print sheet): bad first layer
2nd print (textured print sheet): bad first layer
3rd print (smooth print sheet): perfect first layer
4th print (textured print sheet): perfect first layer
5th print (textured print sheet): bad first layer
6th print (smooth print sheet): perfect first layer

This is weird and I don't get why the first layer results are so inconsistent with the textured print sheet. I cleaned it completely with IPA...

bkerler, 2023-12-21:
Not 100% sure on that but it seems there is an issue with the layer height / loadcell. Was the nozzle clean on each try ? Do you use socks or any modifications ? Otherwise I'd recommend to reset the eeprom if something has been messed up and retry.

sascape, 2023-12-22:
...unbelievable, I can't reproduce the bug without the silicone socket anymore. 

Three tries and the first layer is perfect every time. I've removed the silicone socket and deactivated it in the hardware settings. I put the socket back on and the bug returns.

But it makes sense. I got the silicone socket at the beginning of November but it took around 2-3 weeks to notice those weird first layer issues (most of the time the print succeeded but the surface finish looked weird afterwards). Around that time 5.1 released so my brain went to that as the cause.

Such a simple solution...
Thank you! This bug frustrated me for weeks.

sascape, 2023-12-22:
Sorry, I haven't used github much in the past.

sascape, 2023-12-22:
Sorry... I didn't want to button smash. I haven 't used github like that before.

I thought the silicone socket would be it but the bug finally returned after a few tries :/

Still, thank you! You gave me hope for around half an hour.

bkerler, 2023-12-22:
Yeah that pretty much sounds like an issue with the loadcell. I don't think it's a fw bug, you better contact support and ask for a replacement :)

sascape, 2023-12-22:
I did a full factory reset and the issue is still there so before testing any further I will contact customer support.
Thank you for your help!

bkerler, 2023-12-22:
You're welcome. Keep us updated so others who have the same issue will know what it was in the end ;)

sascape, 2024-01-12:
I sent the customer support this thread and they agreed that it could be the loadcell. Within a week I had the new one for free and I finally had the time to install it today.
What can I say, it worked like a charme, yay :)

closed 2024-01-12 by sascape