prusa3d/Prusa-Firmware-Buddy #3626: [BUG] <Enter comprehensive title>

Cbokar, 2023-12-20:
After some time of printing, 1 hour or so, there are no more temperatures shown under: settings— info —sensors: heatbreak and boardtemperature are shown as: not initialized. 
After a reset the temps are shown properly again.
Printer type - Mk4, factory assembled. FW 5.1.2 slicer 2.7.1

No matter what gcode file, gcode or bgcode, no matter printed from usb directly ore send via connect
    
At the beginning of a print:
https://  github.com/prusa3d/Prusa-Firmware-Buddy/assets/153955662/d85585a4-659c-44b9-a9bc-0f7c78b12a98

At the end of the print.
[image]

[image]

Cbokar, 2023-12-20:
https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/153955662/23b69871-41f0-4cf6-a465-8b49b69d4678

If slowed down. One sees, that each reading is switching from not initialized to the correct reading.
Somehow, after a while only those two values are stuck: not initialized?

bkerler, 2023-12-20:
Could be related to issue #3554

bkerler, 2023-12-21:
I can confirm the same behaviour on my MK4. So clearly a bug.

H0Z0, 2024-04-17:
FW 6.0.0 still has this phenomenon. By the way, not only the board and heatbreak temperatures are not displayed ("not initialized"), but the print fan and heater fan are also N/A in this case. If the board and heatbreak temperatures sometimes show up, it also shows the fans well.

closed 2024-07-09 by github-actions[bot]