prusa3d/Prusa-Firmware-Buddy #3628: [BUG] <Enter comprehensive title>

kldmurshed, 2023-12-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

MK4

5.1.2

Original

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  After updating the firmware 5.1.2 the screen freezes on the Prusa Screen Bootloader 2.3.4

**How to reproduce**
Every time I boot the screen

**Expected behavior**
Should load the printer main screen

[image]

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/869951/b91e6d3e-98d1-4fc8-a9e3-67130f47fd42

danopernis, 2024-09-02:
This seems like it is related to #3333 and #3629

@kldmurshed can you try newest firmware v6.1.2 which contains update to the bootloader?

Prusa-Support, 2024-11-07:
This issue should be very closely related to #3333 and #3629 indeed, and both are closed as solved.
The issue has been silent for a long time and is supposedly fixed by firmware 6.1.2, which came with bootloader upgrade (2.3.5).

Michele Moramarco
Prusa Research

closed 2024-11-07 by Prusa-Support