prusa3d/Prusa-Firmware-Buddy #3629: [BUG] Printer Main screen not loading after firmware update 5.1.2

kldmurshed, 2023-12-20:
MK4

5.1.2

Original

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  After updating the firmware 5.1.2 the screen freezes on the Prusa Screen Bootloader 2.3.4

**How to reproduce**
Every time I boot the screen

**Expected behavior**
Should load the printer main screen

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/869951/59104efa-ff14-40c9-ba38-bf4061f625de

[image]

bkerler, 2023-12-20:
Which fw version did you upgrade from ? It might resolve the issue if you downgrade to an older bootloader (5.1.0) and then retry to upgrade.

Tobiyashi, 2023-12-21:
I had the same issue with the previous firmware.
It helped for me to pull the usb stick after the orange bar reached the middle and stopped.
The printer then asked for a stick with the bbf file and i inserted the stick again.
The flashing then continued as expected.

kldmurshed, 2023-12-22:
> Which fw version did you upgrade from ? It might resolve the issue if you downgrade to an older bootloader (5.1.0) and then retry to upgrade. Also make sure you use a different usb stick just to make sure, some people have issues with the prusa usb stick.

I have attempted to upgrade from the previous version 5.1.0 multiple times, even utilizing a new USB stick. Despite several retries and a few instances of downgrading and upgrading, I have not achieved success.

bkerler, 2023-12-22:
Did you try downgrading to a much older firmware such as 4.4.x ?

bkerler, 2023-12-22:
Another possibility for flashing the firmware would be using prusaslicer and a serial usb cable afair

Area5142, 2023-12-30:
I'm encountering the same issue with firmware version 5.1.2. When I initiate the upgrade process using the "MK4_MK3.9_firmware_5.1.2.bbf" file on a Prusa memory stick, the printer starts upgrading the bootloader and then the firmware. However, it never reaches the main menu. Pressing the reset button or cycling the power does not resolve the issue.

Upon downgrading to firmware version 5.1.0, I was able to access the main menu again. This suggests that the problem is specifically with firmware version 5.1.2. Notably, even though the bootloader was keept as version 2.3.4 during the downgrade process, the printer still functioned normally with firmware 5.1.0.

nils91, 2024-01-04:
Seeing something kinda similar on the MIN, but the behaviour is a bit weird. Already posted on the forum: 
https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/mini-wont-boot-to-menu-from-cold/

> "Since i´ve flashed the 5.1.2 update onto my printer, i have a very weird problem, where the printer won´t boot fully. It will load the firmware (bar goes to 100%), it completes the short fan self test at startup (fans stop spinning), but it w´ont beep and it w´ont go to the menu. An immediate reboot doesn´t fix it, but if i leave the printer in that state for at least a couple minutes it will boot fully, and as far as i can tell, seems fine and prints.
> 
> If a USB stick with the 5.1.2 firmware is connected at cold boot, it will reflash that, but still gets stuck when booting. But same as above, if i leave it for a bit, then reboot, it boots to menu as expected without reflashing."

I think i´ll downgrade for now...

kikolobo, 2024-01-15:
I have the same issue on an MK4. Downgrading work. Downgrading to an older version and then upgrading did not worked for me. Still hangs. Awaiting new fw with fix.

dukea42, 2024-01-15:
I'll add one more to hopefully raise this priority.   If my MK4 needs to restart after a 5.1.2 upgrade (power loss, manual power cycle, reset button, etc.), it hangs on the boot screen.  Forced to reinstall 5.1.0 to complete a full boot to main screen.  If I'm stubborn and try powering off and on a bunch (dozens of times) - it may eventually boot as expected with 5.1.2.

FreeBSDGeek, 2024-01-15:
I hate ot just add a me to but I am seeing the same thing.  
New MK4, Tried 3 different USB sticks, made sure they where FAT32, downloaded the firmware a few times, but no luck.
5.1.0 installs fine but 5.1.2 hangs after the update at the bootloader screen.

khenderick, 2024-01-17:
I have the same thing. Been in contact with support for the issue and we tried various things such as factory reset etc. Sometimes it helps when I power down (cut power) after it flashed (and is stuck on 100%), but not always.

One thing that support said should not be happening is that the bootloader flashes non-interactively. Apparently the bootloader should always ask before flashing if there's a firmware file found on the USB stick. Is it interactive with any of you?

They also suggested to change the battery on the motherboard, but I didn't have a CR1220. Anybody here already tried whether replacing the battery helped?

neslekkim, 2024-01-17:
Regarding changing the battery, what data are we loosing then?, need to do full calibration again? (Not so bad on the MK4, but it's tedious on the XL)

khenderick, 2024-01-17:
Went to buy a new battery, and unfortunately replacing the battery didn't work.

FreeBSDGeek, 2024-01-17:
My Bootloader does not prompt either.  If it finds a firmware, it just starts the firmware flash.  I thought that was normal, glad to hear it is not but mine is doing the same thing.

I worked with support last night.  We tried a number of things and nothing solved it.  In the end we flashed 5.1.0 in (that one works) and he was going to pass the problem off to the MK4 firmware team.  I haven't heard back yet.  He didn't mention the battery but we did to a complete factory reset as well as disconnect all the motors and Nextruder and try it then.  Nothing fixed the 5.1.2 hang yet.

kikolobo, 2024-01-17:
This is a bootloader bug or firmware bug. It won’t work until they fix the FW. My guess is that there are some components in the wild that are slightly different in our control boards and they are not compatible with the BL.  But then again this is just a guess. It might be many things, but for sure it is a software/firmware issue 

there’s not much we can do until they fix it.

FreeBSDGeek, 2024-01-17:
I agree, I think it is something in the Bootloader but it must be an interaction of the Bootloader and something in the firmware.    5.1.2 is the only one I see the problem with. With the same bootloader (2.3.4) I have tried firmware 5.1.0, 5.0.0, 4.7.3 and they all work fine.

It would be great if we could move back to an older bootloader to test but it seems that once the bootloader if updated, it stays.

kikolobo, 2024-01-17:
Sounds right. An interaction of the BL with the FW… 

it might be possible to downgrade the BL using an older firmware that has a boot loader update in it.. but it’s a guess. 

I think we should wait for an update since the cycle of new releases should be close.

kikolobo, 2024-01-27:
Any news if the latest firmware fixes this for everyone? I am in the middle of a large job and can't update for now to check.

kikolobo, 2024-01-29:
Tried the latest 5.1.3. issue still persists... Tried downgrading to an older firmware with boot loader downgrade.. The flashing went smooth but once trying to go back to the latest firmware the issue came back again. Guess has something to do with the UART upgrade process.

larsinka, 2024-01-29:
I have the same issue, printer doesn't boot

FreeBSDGeek, 2024-01-30:
Same issue still as well.  I tried upgrading to the new 5.1.3 and it still is stuck trying to boot.  I see the progress bar move across to 100% and then it just says locked up at that point.  The 5.1.3 firmware didn't solve the issue.

kikolobo, 2024-01-31:
Here’s my solution. Factory Reset with “clear all settings and flash firmware with USB” or a similar options.   I did that and the 5.1.3 flashed fine for me.

khenderick, 2024-01-31:
@kikolobo is that a permanent fix? I also did that with the 5.1.2, but it still gets stuck every now and then. Restarting the printer a dozen times usually gets it to boot up. If you leave the firmware on the USB stick, does it automatically flash the firmware again every time the printer starts?

kikolobo, 2024-01-31:
Don’t know if it’s permanent. So far so good. We’ll see what happens on the next update.

it does not keep loading after the update.

kldmurshed, 2024-01-31:
> Here’s my solution. Factory Reset with “clear all settings and flash firmware with USB” or a similar options. I did that and the 5.1.3 flashed fine for me.

This solution worked for me. Thank you for bringing this up.

kikolobo, 2024-02-01:
I spoke to soon 😊 Upon a reboot the screen froze again 😕

theloukou, 2024-02-07:
> Seeing something kinda similar on the MINI+, but the behaviour is a bit weird. Already posted on the forum: https://forum.prusa3d.com/forum/hardware-firmware-and-software-help/mini-wont-boot-to-menu-from-cold/
> 
> > "Since i´ve flashed the 5.1.2 update onto my printer, i have a very weird problem, where the printer won´t boot fully. It will load the firmware (bar goes to 100%), it completes the short fan self test at startup (fans stop spinning), but it w´ont beep and it w´ont go to the menu. An immediate reboot doesn´t fix it, but if i leave the printer in that state for at least a couple minutes it will boot fully, and as far as i can tell, seems fine and prints.
> > If a USB stick with the 5.1.2 firmware is connected at cold boot, it will reflash that, but still gets stuck when booting. But same as above, if i leave it for a bit, then reboot, it boots to menu as expected without reflashing."
> 
> I think i´ll downgrade for now...

Having the same issue on MINI with last firmware. Loading bar at 100% and just freezes. Bought the printer 6 days ago, so no idea about behavior with other FW versions.
Mine has done this EVERY time i turned it on (which is 4 in total, i haven't printed a lot, but still 100%), besides first boot after build (had not upgraded FW yet then thought).
Haven't figured out what exactly makes it boot eventually. Tried reseting, removing USB, fully pulling the PSU plug. at some point in just boots.
Seems like nothing will ever touch an ATmega and Marlin1 for stability...

bkerler, 2024-02-07:
@theloukou this is a clear indication that there is something wrong in reading the eeprom. This happens if you move from very old to latest firmware versions as it seems. Try to downgrade again to the original firmware and then update the firmware version by version.

khenderick, 2024-02-07:
It also happens if you update version by version. Since PrusaSlicer embeds the latest firmware version in the gcode so the printer warns about it, I always update the printer's firmware right away.

theloukou, 2024-02-08:
> @theloukou this is a clear indication that there is something wrong in reading the eeprom. This happens if you move from very old to latest firmware versions as it seems. Try to downgrade again to the original firmware and then update the firmware version by version.

This does not make any sense to me whatsoever?
If there was a coding issue with reading the EEPROM (whether it be different structure, or whatever), that would have to happen all the time, and never (eventually)boot. Booting can still happen, just havent figured out what is the actual sequence to do that.
I hope they find the root and fix it soon. I was already doubting buying a MINI, now i'm doubling down...

kikolobo, 2024-02-08:
I don’t know what the root cause is but it is annoying.  Hope they fix it soon. And I am sure they will.

queaker, 2024-02-12:
I have the same issue with my xl2t. after upgrading from 5.1.0 to 5.1.2 the device isn't booting any more....

bkerler, 2024-02-13:
> This does not make any sense to me whatsoever? If there was a coding issue with reading the EEPROM (whether it be different structure, or whatever), that would have to happen all the time, and never [...]

On boot, the configuration store is being read from eeprom, which resides on the loveboard. Basically it could be any initialisation that fails that keeps the bootloader from booting, but if that happens on an upgrade, it's much likely either the eeprom or the esp causing this specific issue. Can you reset to factory settings and see if the behaviour changes on the next start ?

Also it's very different depending on MINI, MK4 and XL and of course if it hangs at 50% or 100%.

bkerler, 2024-02-13:
> I have the same issue with my xl2t. after upgrading from 5.1.0 to 5.1.2 the device isn't booting any more....
Did you upgrade via usb stick or via prusaslicer ? Can you flash 5.1.0 via prusaslicer and then upgrade to 5.1.2 via prusaslicer ?

bkerler, 2024-02-13:
@kikolobo after you did the reset, did you change some settings before it was hanging again ? and if yes, which settings did you change ?

theloukou, 2024-02-13:
> On boot, the configuration store is being read from eeprom, which resides on the loveboard. Basically it could be any initialisation that fails that keeps the bootloader from booting, but if that [...]

I am having this issue on a MINI+ (as @nils91 is), so no loveboard, no ESP plugged in either. Also it didnt happen the first time after an upgrade (the upgrade booted just fine actually). It happens EVERY time (needless to say that there is no update happening everytime...).
If you "blame" the EEPROM and initialization sequence, then either there is something in that sequence that is not optimized for all printers, or there is some bug/issue between MINI's Buddy board and the unified firmware.
Unfortunately i have no experience with Marlin2, let alone the STM32 firmware, so there is no hope and time for me to even start looking into it on my side.

bkerler, 2024-02-15:
@theloukou As this normally doesn't happen (and it's impossible to find that bug/issue) I think it would require you to send in the board so the issue can be debugged using Jtag. Please ask Prusa support if that would be possible.

khenderick, 2024-02-15:
I've been in contact with support regarding this issue and they didn't ask to send the board back. However, I stopped getting updates on that ticket, so not sure on the state.

Note however that you can just update to an older version and the new bootloader will (without issue) boot in that version. It's only the new 2.3.4 bootloader in combination with >= 5.1.2 that gives the issue. 2.3.4 + 5.1.0 is fine.

queaker, 2024-02-15:
> > I have the same issue with my xl2t. after upgrading from 5.1.0 to 5.1.2 the device isn't booting any more....
> 
> @queaker Did you upgrade via usb stick or via prusaslicer ? Can you flash 5.1.0 via prusaslicer and then upgrade to 5.1.2 via prusaslicer ?

I did a downgrade to 5.1.0 via usb-drive. after the downgrade the printer starts again. but now i have to confirm every print because there is a new firmware.

i tried the upgrade again yesterday. but this time from prusa connect. another attempt to upgrade to 5.1.2 again results in the printer no longer starting.

bkerler, 2024-02-15:
@queaker I assume you didn't get any warning or error on upgrading ? My guess is that either the bootloader is never really upgraded (or something prevents the upgrade) or that a specific eeprom setting might prevent booting. If you downgrade to a version without bootloader upgrade and you upgrade the firmware I assume you aren't hitting this issue ?

queaker, 2024-02-15:
> @queaker I assume you didn't get any warning or error on upgrading ? My guess is that either the bootloader is never really upgraded (or something prevents the upgrade) or that a specific eeprom [...]

I did the firmware downgrade as follows.

1. i copied the old firmware (5.1.0) version on the usb-stick
2. I pressed and held the rotary knob at startup to force a firmware install
3. I watched the printer during the firmware downgrade.

i don't know whether the bootloader is also downgraded. should i start the downgrade in another way?

bkerler, 2024-02-15:
@queaker to be honest, I haven't looked too deep into the bootloader process, that's clearly on my todo to fully understand how it works. What you could try is to use the prusaslicer and the usb connection instead to flash the firmware (if that does change anything). Having a look at the firmware [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/b3ee47aa573f4d9b4af82d7a6963f43ab37ccc08/src/gui/MItem_tools.cpp#L221), only the hard reset does wipe the previous firmware, whereas both the soft reset and the hard reset do wipe the eeprom values. A hard reset and then to flash the 5.1.2 (ideally via prusaslicer) would definitely be worth a try.

bkerler, 2024-02-15:
@khenderick
Mhmm so it clearly is something that was introduced with 5.1.2 as the bootloader is the same in your case.

khenderick, 2024-02-16:
> @khenderick Mhmm so it clearly is something that was introduced with 5.1.2 as the bootloader is the same in your case.

Indeed. And support told me also it was highly unexpected that the bootloader didn't interactively ask me to flash the firmware (even if it was a downgrade, or it was the same version as was already in place). Maybe that helps your insights in the issue.

queaker, 2024-03-06:
On 6.0.0-alpha the machine is still not booting after an upgrade from 5.1.0.

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/4228208/c000454f-edc0-4ad0-80da-70798174f1af

bkerler, 2024-03-06:
@FreeBSDGeek Thanks for letting us know.

queaker, 2024-03-06:
i contacted customer support today with the case. they guided me through a hard reset. since then the printer boots reliably even after an update.

https://help.prusa3d.com/article/factory-reset-mk4-xl_421206

theloukou, 2024-03-06:
I decided to try stuff today (MINI+). After a (very frustrating) 2:30 hour chat with 2 online support guys (1st guy kept asking for various stuff until his shift ended?!?!?), I ended up doing various things for lols: reformatting the usb, reflashing about 10 times, hard resets, downgraded to 4.4.1, updating to 5.1.0, then 5.1.2 gradually.
Every other FW version works fine, except 5.1.2.
It is super annoying that eveyone's observations point to just version 5.1.2 but noone from Prusa devs seems to ask for logs, dumps, or SOMETHING to try to figure it out...

aa956, 2024-03-07:
Got the same issue with the new mk4 that was shipped with 5.1.2 and was in this bootloop.

Downgrade to 5.1.0 helped, thanks.

bkerler, 2024-03-07:
I still wonder what might cause this. If it still stays after the hard reset, the eeprom likely isn't the cause. When updating the firmware, was the wifi adapter connected and if yes, can someone try to remove it before updating ? Also I saw that at the video of @queaker the scrollbar appeared, hangs at 50%, still shows the bootloader version message but then the bootloader version disappears. Is that the same for all others ?

theloukou, 2024-03-07:
@bkerler I don't have ESP module installed at all.
@aa956 downgrading is NOT a solution. It is a remedy. A compromise of loosing some stuff, and getting back normal booting

bkerler, 2024-03-07:
@theloukou does it behave the same way if you install the esp module and try upgrading ?

nils91, 2024-03-07:
> I still wonder what might cause this. If it still stays after the hard reset, the eeprom likely isn't the cause. When updating the firmware, was the wifi adapter connected and if yes, can someone [...]

On the MINI+ (at least for me) its slightly different in that it goes to 100%. I can also get it to boot sometimes if I power cycle the MINI often enough, although that has a very low chance of success.
Other than that, the symptoms are similar, 5.1.0 works and 5.1.2 (most of the time) does not, no matter what combination of upgrading/downgrading/resetting I try.

Prusa-Support, 2024-03-14:
Thanks for reporting. Our developers are still working on the reproducibility of the issue.

For now please, if a [hard reset via the menu](https://help.prusa3d.com/article/factory-reset-mk4-xl_421206) doesn't work, please try it via bootloader as it follows.

- On bootloaders 2.1.0 and higher, you plug in the USB flash drive with one only .bbf firmware file - you can try with the FW 5.1.2 in this case.
- When the firmware flashing screen appears and the countdown starts, you just hold the knob on the `FLASH` option for 3-4 seconds.
- A `DON'T PANIC` screen will appear. Now you can complete the process by hitting the `ERASE` command.

[image]

Michele Moramarco
Prusa Research

Area5142, 2024-03-15:
I tried the Hard-Reset method with version FW 5.1.3 on my MK4 as suggested by @Prusa-Support and it seems to work.
The downside is that I lost all printing statistics and had to run the setup wizard again.

Prusa-Support, 2024-03-26:
Glad to hear that it helped.
I hope the problem didn't show up again in the last two weeks and I wonder if that was helpful for other users too.

Michele Moramarco
Prusa Research

aa956, 2024-03-28:
For assembled MK4 ordered and received in February 2024 with 5.1.2 firmware preinstalled that was showing this error on first startup after unpacking:

* Was getting the same error with both 5.1.2 reinstalled and 5.1.3 firmware, was starting correctly when downgraded to 5.1.0.
* Hard reset via bootloader on 2024-03-22 and installed firmware 5.1.3
* At the moment no signs of this problem anymore. Will update this issue if or when this will reoccur.

nils91, 2024-03-28:
On my MINI unfortunately that hasnt been a permanent fix. The hard reset worked initially, but the problem came back after a few days.

theloukou, 2024-03-28:
Not fix on my MINI as well.

kldmurshed, 2024-03-30:
I believed the problem would be resolved in version 6.0.0-RC1. However, I'm still encountering the same issue with my machine, and I'm still on version 5.1.0. I'm requesting @Prusa-Support to address this issue in the upcoming final release of 6.0.0.

[23 more replies]

closed 2024-09-14 by Prusa-Support